FreeOpcUa/opcua-asyncio #692: message serialization is slow

lovasoa, 2021-10-20:
Hello, and thank you for this great library !

  We use it on embedded hardware in an environment where performance matters, and after some profiling, we noticed that around 10% of the run time of our program is spent in the single function [`ua::ua_binary::struct_to_binary`](https://github.com/FreeOpcUa/opcua-asyncio/blob/b19881b91a3efe8f10eff9dfff213e822812e734/asyncua/ua/ua_binary.py#L227).
  
  I have a few ideas to improve performance of serialization in asyncua, and before I start writing code, I would like to know if you would be ready to merge pull requests that slightly increase the code complexity for performance gains.
  
  I am mainly thinking about two things: 
  
  - generating the serialization functions ahead of time instead of processing each dataclass field type every time an object instance is serialized 
  - reducing the amount of small bytes() object allocation and concatenation by pre-allocationg buffers and using struct.pack_into on them.

lovasoa, 2021-11-05:
[image]

lovasoa, 2021-11-05:
I'm going to make a new pr if that's ok

oroulet, 2021-11-05:
improvments always welcome