FreeOpcUa/opcua-asyncio #702: Increase the usage of augmented assignment statements

elfring, 2021-10-31:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

```diff
diff --git a/asyncua/crypto/security_policies.py b/asyncua/crypto/security_policies.py
index 4fbf195..44309cc 100644
--- a/asyncua/crypto/security_policies.py
+++ b/asyncua/crypto/security_policies.py
@@ -159,7 +159,7 @@ class Cryptography(CryptographyNone):
             rem = block_size - rem
         data = bytes(bytearray([rem % 256])) * (rem + 1)
         if self.Encryptor.encrypted_block_size() > 256:
-            data = data + bytes(bytearray([rem >> 8]))
+            data += bytes(bytearray([rem >> 8]))
         return data
 
     def min_padding_size(self):
@@ -599,7 +599,7 @@ class SecurityPolicyBasic256(SecurityPolicy):
             self.symmetric_cryptography.prev_key_expiration = self.symmetric_cryptography.key_expiration
 
         # convert lifetime to seconds and add the 25% extra-margin (Part4/5.5.2)
-        lifetime = lifetime * 1.25 * 0.001
+        lifetime *= 1.25 * 0.001
         self.symmetric_cryptography.key_expiration = time.time() + lifetime
         self.symmetric_cryptography.Verifier = VerifierAesCbc(sigkey)
         self.symmetric_cryptography.Decryptor = DecryptorAesCbc(key, init_vec)
diff --git a/asyncua/ua/uatypes.py b/asyncua/ua/uatypes.py
index 9897650..b9ce4b8 100644
--- a/asyncua/ua/uatypes.py
+++ b/asyncua/ua/uatypes.py
@@ -242,7 +242,7 @@ class _MaskEnum(IntEnum):
         while n:
             if n & 0x1:
                 yield pos
-            n = n // 2
+            n //= 2
             pos += 1
 
 
```

oroulet, 2021-10-31:
you are welcome. Make a PR

closed 2023-03-29 by schroeder-

elfring, 2023-03-29:
:thought_balloon: How does the issue status indication “not planned” fit to [my pull request for which a commit was merged on 2021-11-1](https://github.com/FreeOpcUa/opcua-asyncio/pull/703 "Convert three statements to the usage of augmented assignments")?