FreeOpcUa/opcua-asyncio #721: Question: Is it possible to create custom objects through a client

liona24, 2021-11-14:
Hey there,
I was wondering whether it is possible to create a custom structure / object by invoking a service as a client and if so which service should I use. Consider f. e. the examples [here](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-custom-structures-amd-enums.py) or [here](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-custom-object.py#L45).

I am not that familiar with all the UA services therefor figured asking some of you more experienced folks is probably the fastest solution. Thanks.

AndreasHeine, 2021-11-14:
https://github.com/FreeOpcUa/opcua-asyncio/blob/77796a37e3e0e368cf82687fdbc95525b084e517/asyncua/client/ua_client.py#L580

you can create nodes from client side but you can only use the types the server already knows!
Service -> NodeManagement Service Set -> https://reference.opcfoundation.org/v104/Core/docs/Part4/5.7.2/

liona24, 2021-11-14:
Got it, though I was kind of hoping that clients could create the types as well.
Anyway thank you for your swift response :)

closed 2021-11-14 by liona24

AndreasHeine, 2021-11-15:
;) yeah, actually your usecase is not what opc ua is designed for ^^ its more about the server exposing his types so the client can explore them without knowing them before connect.