FreeOpcUa/opcua-asyncio #727: Can't connect

NizarMhatli, 2021-11-18:
I set up OPC-UA server on Omorn NX102 controller , i can access it using UAexpert but when trying to connect to the server using this library i get the following error:
INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
INFO:asyncua.client.ua_client.UaClient:create_session
WARNING:asyncua.uaprotocol:Received an error: ErrorMessage(Error=StatusCode(value=2153054208), Reason=None)
CRITICAL:asyncua.client.ua_client.UASocketProtocol:Received an error: ErrorMessage(Error=StatusCode(value=2153054208), Reason=None)
ERROR:asyncua.client.ua_client.UASocketProtocol:Exception raised while parsing message from server
Traceback (most recent call last):
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 167, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 79, in _process_received_data
    self._process_received_message(msg)
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 103, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 169, in _call_callback
    raise ua.UaError(f"No request found for request id: {request_id}, pending are {self._callbackmap.keys()}, body was {body}") from ex
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([2]), body was "The security policy does not meet the requirements set by the server."(BadSecurityPolicyRejected)
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
WARNING:asyncua.client.ua_client.UaClient:disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "C:/Users/k159/Desktop/pythonProject_OPC_UA/main.py", line 35, in <module>
    asyncio.run(main())
  File "C:\Users\k159\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\k159\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "C:/Users/k159/Desktop/pythonProject_OPC_UA/main.py", line 16, in main
    async with Client(url=url) as client:
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\client.py", line 68, in __aenter__
    await self.connect()
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\client.py", line 247, in connect
    await self.create_session()
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\client.py", line 370, in create_session
    response = await self.uaclient.create_session(params)
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 297, in create_session
    data = await self.protocol.send_request(request)
  File "C:\Users\k159\Desktop\pythonProject_OPC_UA\venv\lib\site-packages\asyncua\client\ua_client.py", line 148, in send_request
    raise ConnectionError("Connection is closed") from None
ConnectionError: Connection is closed

AndreasHeine, 2021-11-18:
you try to connect with the wrong settings -> BadSecurityPolicyRejected

NizarMhatli, 2021-11-18:
im new to OPC-UA so here is a preview of the setup in Uaexpert and the code that i wanted to try 
[image]
[image]
)

AndreasHeine, 2021-11-18:
your python client is not configured for sign & encrypt! you should look in our examples folder!

closed 2022-02-20 by AndreasHeine