FreeOpcUa/opcua-asyncio #729: new version of asyncua produces decoding error when variables is read from asyncua server

thunderbug1, 2021-11-18:
### Please read, before you post!

After upgrading I encountered an issue where ua expert gets BadDecodingError when reading any variable from the asyncua server (folders work fine).
This did not happen with an older version of asyncua.
Here are screenshots of ua expert 

asyncua==0.9.14    
[image]

asyncua==0.9.91 (exact same code as before)
[image]

everything was run with Python 3.8.10

I didn't have time to build a minimal example yet.

oroulet, 2021-11-18:
What kind of server?

oroulet, 2021-11-18:
I tested all kind of possible variables with asyncua server and uaexpert and prosys client as client and everything seems fine. must be something special with your setup. or something here that makes it work

oroulet, 2021-11-18:
last asyncua version is 0.9.92, try to upgrade first. I thing there was a bad bug in 0.9.91

thunderbug1, 2021-11-19:
@oroulet oh yes thank you, with the version 0.9.92 the bug seems to be gone.

closed 2021-11-19 by thunderbug1