FreeOpcUa/opcua-asyncio #734: load_data_type_definitions() Fails with Enumeration within class definition,

robbudge, 2021-11-24:
Within my OPCua Server we have a DataType with the following format.

HMI_CMD: REAL
PLC_CMD: REAL
Valve_Actual:REAL
Mode:Burkert_ACV_Mode
Status: Burkert_ACV_Status

The Mode & Status Enumerators are being detected and built correctly
as shown in the log

> class Burkert_ACV_Modes(IntEnum):
> 
>     '''
>     Burkert_ACV_Modes EnumInt autogenerated from EnumDefinition
>     '''
> 
>     Auto = 0
>     Closed = 1
>     Open = 2
>     Manual_Pos = 3
> 
> INFO:asyncua.ua.uatypes:registring new enum: Burkert_ACV_Modes NodeId(Identifier='|tprop|WAGO 750-8101 PFC100 2ETH.PLC.Burkert.ACV_Modes', NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>) <enum 'Burkert_ACV_Modes'>
> WARNING:asyncua.common.structures104:renamed Burkert.ACV_Status to Burkert_ACV_Status due to Python syntax
> INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier='|tprop|WAGO 750-8101 PFC100 2ETH.PLC.Burkert.ACV_Status', NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>) Burkert_ACV_Status OptionSet=False
> INFO:asyncua.common.structures104:Registering data type NodeId(Identifier='|tprop|WAGO 750-8101 PFC100 2ETH.PLC.Burkert.ACV_Status', NamespaceIndex=4, NodeIdType=<NodeIdType.String: 3>) QualifiedName(NamespaceIndex=4, Name='Burkert.ACV_Status')
> ERROR:asyncua.common.structures104:Offline - EnumField(Value=7, DisplayName=LocalizedText(Locale=None, Text='Offline'), Description=LocalizedText(Locale=None, Text=None), Name='Offline') False 
> 
> class Burkert_ACV_Status(IntEnum):
> 
>     '''
>     Burkert_ACV_Status EnumInt autogenerated from EnumDefinition
>     '''
> 
>     Auto = 0
>     Closed = 1
>     Open = 2
>     Disabled = 3
>     Manual = 4
>     Fault = 5
>     Locked_Out = 6
>     Offline = 7
> 

When the actual data class is being built.

> @dataclass
> class Burkert_ACV_DUT:
> 
>     '''
>     Burkert_ACV_DUT structure autogenerated from StructureDefinition object
>     '''
> 
>     data_type = ua.NodeId.from_string("ns=4;s=|tprop|WAGO 750-8101 PFC100 2ETH.PLC.Burkert.ACV_DUT")
> 
>     HMI_CMD: ua.Float = 0
>     PLC_CMD: ua.Float = 0
>     Valve_Actual: ua.Float = 0
>     Mode: ua.Enumeration = ua.Enumeration()
>     Status: ua.Enumeration = ua.Enumeration()
> Traceback (most recent call last):
>   File "/Users/robertbudgen/PycharmProjects/OPCua_Async_1/venv/lib/python3.8/site-packages/asyncua/common/structures104.py", line 245, in _generate_object
>     exec(code, env)
>   File "<string>", line 4, in <module>
>   File "<string>", line 15, in Burkert_ACV_DUT
> **AttributeError: module 'asyncua.ua' has no attribute 'Enumeration'**

Im suspecting this could be a general OPCua Issue and not OPCua-asyncio specific.
when monitoring with UaExpert, the 2 Enumerated variables are excluded.

oroulet, 2021-11-25:
ok. looks like a bug/limitation. the default value should have been something else than enumeration()

oroulet, 2021-11-25:
I checked and we have a test for that particular situation. Are you sure you have latest released version of opcua-asyncio?
If you do there i something special in your case...

schroeder-, 2023-03-29:
Should be fixed in current version, otherwise reopen issue.

okapies, 2023-07-13:
Hi, @schroeder-.  I got the same error when try to connect to the [reference server](https://github.com/OPCFoundation/UA-.NETStandard/tree/master/Applications/ReferenceServer):

```
asyncua.common.structures104 - ERROR - Failed to execute auto-generated code from UA datatype: 

@dataclass
class ScalarStructureDataType:

    '''
    ScalarStructureDataType structure autogenerated from StructureDefinition object
    '''

    data_type = ua.NodeId.from_string('''ns=3;i=1078''')

    BooleanValue: ua.Boolean = True
    SByteValue: ua.SByte = ua.SByte(0)
    ByteValue: ua.Byte = ua.Byte(0)
    Int16Value: ua.Int16 = ua.Int16(0)
    UInt16Value: ua.UInt16 = ua.UInt16(0)
    Int32Value: ua.Int32 = ua.Int32(0)
    UInt32Value: ua.UInt32 = ua.UInt32(0)
    Int64Value: ua.Int64 = ua.Int64(0)
    UInt64Value: ua.UInt64 = ua.UInt64(0)
    FloatValue: ua.Float = ua.Float(0)
    DoubleValue: ua.Double = ua.Double(0)
    StringValue: ua.String = ua.String()
    DateTimeValue: ua.DateTime = datetime.utcnow() # type: ignore
    GuidValue: ua.Guid = uuid.uuid4()
    ByteStringValue: ua.ByteString = b''
    XmlElementValue: ua.XmlElement = field(default_factory=ua.XmlElement)
    NodeIdValue: ua.NodeId = field(default_factory=ua.NodeId)
    ExpandedNodeIdValue: ua.ExpandedNodeId = field(default_factory=ua.ExpandedNodeId)
    QualifiedNameValue: ua.QualifiedName = field(default_factory=ua.QualifiedName)
    LocalizedTextValue: ua.LocalizedText = field(default_factory=ua.LocalizedText)
    StatusCodeValue: ua.StatusCode = field(default_factory=ua.StatusCode)
    VariantValue: ua.Variant = field(default_factory=ua.Variant)
    EnumerationValue: ua.Enumeration = field(default_factory=ua.Enumeration)
    StructureValue: ua.ExtensionObject = field(default_factory=ua.ExtensionObject)
    NumberValue: ua.Number = field(default_factory=ua.Number)
    IntegerValue: ua.Integer = field(default_factory=ua.Variant)
    UIntegerValue: ua.UInteger = field(default_factory=ua.Variant)
')
Traceback (most recent call last):
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 299, in _generate_object
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 34, in ScalarStructureDataType
AttributeError: module 'asyncua.ua' has no attribute 'Enumeration'
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - module 'asyncua.ua' has no attribute 'Enumeration'')
Traceback (most recent call last):
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/opcua_client-0.8.4-py3.10.egg/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/opcua_client-0.8.4-py3.10.egg/uaclient/uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/sync.py", line 95, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/sync.py", line 53, in post
    return futur.result()
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/client/client.py", line 833, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 474, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type, log_fail=log_ex)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 299, in _generate_object
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 34, in ScalarStructureDataType
AttributeError: module 'asyncua.ua' has no attribute 'Enumeration'
```

schroeder-, 2023-07-14:
Should be an easy fix we just make an alias for `Enumeration = ua.Int32 `

For a workaround add this before connecting:
`ua.Enumeration = ua.Int32`

okapies, 2023-07-14:
Thanks. I also faced another issue for `Number` after using your workaround, and failed to handle `StructureField`.

```
uaclient.uaclient - INFO - Connecting to opc.tcp://localhost:62541 with parameters None, None, None, None')
asyncua.common.structures104 - WARNING - renamed 3DVector to _3DVector due to Python syntax')
asyncua.common.structures104 - WARNING - renamed 3DCartesianCoordinates to _3DCartesianCoordinates due to Python syntax')
asyncua.common.structures104 - WARNING - renamed 3DOrientation to _3DOrientation due to Python syntax')
asyncua.common.structures104 - WARNING - renamed 3DFrame to _3DFrame due to Python syntax')
asyncua.common.structures104 - ERROR - Unknown datatype for field: StructureField(Name='VariantDataType', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=1709, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:UserScalarValueDataType, please report')
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - Unknown datatype for field: StructureField(Name='VariantDataType', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=1709, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:UserScalarValueDataType, please report')
Traceback (most recent call last):
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/opcua_client-0.8.4-py3.10.egg/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/opcua_client-0.8.4-py3.10.egg/uaclient/uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/sync.py", line 95, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/sync.py", line 53, in post
    return futur.result()
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/home/user/.pyenv/versions/3.10.12/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/client/client.py", line 833, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 474, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type, log_fail=log_ex)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 296, in _generate_object
    code = make_structure_code(data_type, name, sdef, log_error=log_fail)
  File "/home/user/opcua-client-gui/venv/lib/python3.10/site-packages/asyncua-1.0.2-py3.10.egg/asyncua/common/structures104.py", line 216, in make_structure_code
    raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")
RuntimeError: Unknown datatype for field: StructureField(Name='VariantDataType', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=1709, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:UserScalarValueDataType, please report
uaclient.mainwindow - WARNING - showing error: %s')
```