FreeOpcUa/opcua-asyncio #760: When I shut down my opc-server, my client neither reported an error nor ran

MansonHua, 2021-12-31:
At first, I wanted the client to automatically reconnect when the server was disconnected.
So I used the server_minimal.

<details>
<summary>server-minimal</summary>
```python
    from asyncua import ua, Server
    from asyncua.common.methods import uamethod
    import logging
    import asyncio
    import sys
    import time

    from asyncua.ua.uatypes import Int32
    sys.path.insert(0, "..")
    QtyPoints = 10
    @uamethod
    def func(parent, value):
        return value * 2
    async def main():
        _logger = logging.getLogger('asyncua')
        # setup our server
        server = Server()
        await server.init()
        server.set_endpoint('opc.tcp://0.0.0.0:4840/freeopcua/server/')

        # setup our own namespace, not really necessary but should as spec
        uri = 'http://examples.freeopcua.github.io'
        idx = await server.register_namespace(uri)

        # populating our address space
        # server.nodes, contains links to very common nodes like objects and root
        myobj = await server.nodes.objects.add_object(idx, 'Monitor')
        myvar = {}
        for i in range(QtyPoints):
            var = await myobj.add_variable(idx, 'var' + str(i), 1 )
            await var.set_writable()
            myvar[i] = var
        countvar = await myobj.add_variable(idx, 'count', "none")
        await server.nodes.objects.add_method(ua.NodeId('ServerMethod', 2), ua.QualifiedName('ServerMethod', 2), func, [ua.VariantType.Int64], [ua.VariantType.Int64])
        _logger.info('Starting server!')
        async with server:
            count = 0
            new_val = 0
            while True:
                await asyncio.sleep(5)
                new_val = new_val + 1
                count = count + 1
                for i in range(QtyPoints):
                    await myvar[i].write_value(new_val)
                if 0 == (count % 100):
                    timeNow = time.time()
                    writevalue = int(count*QtyPoints)
                    writevalue = str(writevalue)+","+str(timeNow)
                    await countvar.write_value(writevalue)```
</details>

This is my client. I subscribe all the nodes of opc-server and write all the changed to .txt.
<details>
<summary>myclient</summary>

```
import datetime
import sys
from asyncua import Client, Node, ua
import asyncio
import aiofile
import time

class SubscriptionHandler():
    async def datachange_notification(self, node, val, data):
        global QtyDataChange
        timeNow = str(time.time())
        try:
            node_name = await node.read_display_name()
            match node_name.Text:
                case 'count':
                    await que.put(timeNow+","+str(node_name.Text)+","+str(val)+","+str(QtyDataChange)+"\n")
                    QtyDataChange = QtyDataChange + 1
                    print("count")
                case _:
                    await que.put(timeNow+","+str(node_name.Text)+","+str(val)+"\n")
                    QtyDataChange = QtyDataChange + 1
                    print("val")
        except BaseException:
            print("error")

async def main():
    global today,oclient
    """
    Main task of this Client-Subscription example.
    """
    # while True:
    async with oclient:
        # ServerInterfaces = await oclient.nodes.objects.get_child("3:ServerInterfaces")
        ServerInterfaces = await oclient.nodes.objects.get_child("2:Monitor")
        node = await ServerInterfaces.get_children()
        await asyncio.sleep(5)
        handler = SubscriptionHandler()
        # We create a Client Subscription.
        subscription = await oclient.create_subscription(500, handler)
        await subscription.subscribe_data_change(node)
        async with aiofile.async_open('out.txt', 'w+') as afp:
            while True:
                result = await que.get()
                print(result)
                await afp.write(result)
                await asyncio.sleep(0)
                print("test")
if __name__ == '__main__':
    if len(sys.argv) == 2:
        url = sys.argv[1]
    else:
        url = 'opc.tcp://127.0.0.1:4840/freeopcua/server/'
    oclient = Client(url, timeout=10)
    que = asyncio.Queue()
    today = (datetime.datetime.now()).strftime('%Y%m%d')
    QtyDataChange = 0
    path = today + ".txt"
    asyncio.run(main())
```
</details>
When i shut down my server, my client neither stopped and reported an error nor keeped running. It just ran without do anything.
I cannot find the cause of this problem.
And i hope to know how to make my client reconnect my server automatically until it connects the server again

oroulet, 2022-01-10:
yes in the example over you need to read a node every x seconds in the main loop and reconnect if any issue. I think it has been asked many times.

schroeder-, 2023-03-29:
See examples/client-reconnect.py

closed 2023-03-29 by schroeder-