FreeOpcUa/opcua-asyncio #762: should use await for connect or not? how to connect from __init__?

ronytesler, 2022-01-04:
why doesn't the file client_to_prosys_cert.py use await?

I want to connect the server when my module loads, so I want to call connect in __init__. But it's not possible to call await from init. What should I do?

oroulet, 2022-01-04:
`__init__()` is not async . the usual way to bypass that issue is to add a method `init()`to your class that is async. I think InternalServer in asyncua does that. t

ronytesler, 2022-01-04:
@oroulet What does this code from client_to_prosys_cert.py work without calling await?
        ```
        client.connect()
        root = client.nodes.root
        objects = client.nodes.objects
        print("childs og objects are: ", objects.get_children())
```

oroulet, 2022-01-04:
it does not work. Looks like that file has never been ported since fork from python-opcua

oroulet, 2022-01-04:
look at client_to_prosys.py. it is correct

ronytesler, 2022-01-04:
ah great, thanks.
so asyncio.run(main()) is the way to call an async function from `__init__`?

oroulet, 2022-01-04:
asyncio.run(main()) is the entry point of you entire async programm. there is not `__init__` in that case. It is a function

ronytesler, 2022-01-04:
Right. But is calling asyncio.run(main()) from `__init__` (in my own package where I do have it) the same as calling it from outside a function?

oroulet, 2022-01-04:
Never call Asuncio.run() from anywhere else than a python main entry file. Read an async tutorial

closed 2022-01-04 by ronytesler