FreeOpcUa/opcua-asyncio #777: Bug in simple-client-server-xml example

danmoser, 2022-01-21:
**Describe the bug** <br />
Trying to run the example "simple-client-server-xml" server triggers the following error:
``` 
simple-client-server-xml$ python3 server.py 
/path/to/simple-client-server-xml/server.py:52: RuntimeWarning: coroutine 'Server.import_xml' was never awaited
  self.server.import_xml(model_filepath)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Traceback (most recent call last):
  File "/path/to/simple-client-server-xml/server.py", line 81, in <module>
    with HelloServer(
  File "/path/to/simple-client-server-xml/server.py", line 61, in __init__
    hellower_say_hello = hellower.get_child("0:SayHello")
AttributeError: 'coroutine' object has no attribute 'get_child'
sys:1: RuntimeWarning: coroutine 'Server.get_namespace_index' was never awaited
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
sys:1: RuntimeWarning: coroutine 'Node.get_child' was never awaited
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

**Version**<br />
Python-Version: 3.9.5
opcua-asyncio Version (e.g. master branch, 0.9): 0.9.92

swamper123, 2022-01-24:
@oroulet was this imported from the sync version? There are no `await`s anywhere?!

oroulet, 2022-01-24:
yes it looks like that example has never been ported. Strange I thought they were all ported now

oroulet, 2022-01-24:
ok that entire example is in a subfolder. I cannot remember ever seeing it. That is why it escaped porting to async

oroulet, 2022-02-20:
https://github.com/FreeOpcUa/opcua-asyncio/pull/814

closed 2022-02-20 by oroulet