FreeOpcUa/opcua-asyncio #778: Calling method - BadNoMatch

sschuhmann, 2022-01-27:
I´m trying to call an OPC-UA method from a method node with the NodeID of ´ns=8;s=4711´. The node is type NodeClass.Method, however, I´m unable to execute the corresponding method. 

Ways that didn´t work:

* Get the method node, call method -> methodid needed
* Get the method node, call with nodeid -> BadNoMatch (The requested operation has not match to return)
* Get the parent node, call with nodeid -> BadNoMatch
 
I have also tried different variations, adding the method description to the nodeid, but this didn´t work as well. 

Is there any way how I can find the method id?

Thank you

AndreasHeine, 2022-01-27:
i assume the nodeid is wrong: ´ns=8;s=4711´ -> ´ns=8;i=4711´ ?

you should use UAExpert to connect to the opc ua server in order to test!
https://www.unified-automation.com/products/development-tools/uaexpert.html

sschuhmann, 2022-01-27:
Thank you for the quick response!

I use UAExpert to experiment with the server, it shows me the same nodeid. 

Also UAExpert allows me to execute the method and receive the response

AndreasHeine, 2022-01-27:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.11.2/

for a ua method call you need:
objectId | NodeId (nodeid from the parent node of that method)
methodId | NodeId (nodeid of the methodnode itself)
inputArguments [] | BaseDataType

https://github.com/AndreasHeine/opcua-tutorial/blob/13f26a962d466c142d42eb3fe9bf53938b7a07d6/client/call_method.py#L52

sschuhmann, 2022-01-27:
Thank you for your comments, I was able to solve it.

In the linked python code you´re using a node object to call the method. In the example files only a string was passed to the method. After switching to a manual translation of nodeids to node objects everything works

swamper123, 2022-01-27:
Case closed :detective:

closed 2022-01-27 by swamper123