FreeOpcUa/opcua-asyncio #779: Server crash with HistoryDict and empty SourceTimestamp

ondrejnovakcvut, 2022-01-27:
**Describe the bug** <br />
When running the server with `HistoryDict` enabled and using a client that does not set SourceTimestamp (tested with UA Expert), function `read_raw_history` fails with `TypeError: '<=' not supported between instances of 'datetime.datetime' and 'NoneType'` when it tries to filter out the values out of range given by function parameters  `starttime` and `endtime`.

**To Reproduce**<br />
1. Run opcua-asyncio server with `HistoryDict` enabled (`server.iserver.history_manager.set_storage(HistoryDict())`)
2. Create a variable, set it writable (`myvar.set_writable()`) and enable its historization (`server.historize_node_data_change(myvar, period=None, count=100)`)
3. Change a value of a variable by client that does not set `SourceTimestamp`
4. Call `server.read_raw_history(numvalues=10)`

**Expected behavior**<br />
I'm not sure what the function should do with records having `SourceTimestamp` equal to `None`. `HistorySQLite` seems to ignore these values (the list only contains values having `SourceTimestamp` set). Maybe there could be a parameter to `read_raw_history` that would tell the function to include these values.

**Version**<br />
Python-Version: 3.9
opcua-asyncio Version):  asyncua==0.9.92 from pypi

oroulet, 2022-01-28:
that dict history stuff was more a code example that something really usefull. fix welcome. It should not be really hard

AndreasHeine, 2022-01-28:
does values without timestamps make sense in a history?

ondrejnovakcvut, 2022-02-03:
> does values without timestamps make sense in a history?

it probably doesn't, but it's handy for debugging

> that dict history stuff was more a code example that something really usefull. fix welcome. It should not be really hard

I'll take a look on it