FreeOpcUa/opcua-asyncio #780: How to load type dictionaries?

deeTEEcee, 2022-01-27:
My main issue is I am subscribed to events but I have data that is encoded in an ExtensionObject and I need to decode it. I believe that something like `load_type_definitions` just handles that magic for me assuming the OPC server has the right structure but I end up run into some issues.

Following this [thread](link), it seems to suggest I should be using `load_type_definitions` with a node one level below the OPC Binary. (In the data types, I see 'XML Schema' and 'OPC Binary' objects which is why I believed it's a pre-1.04 schema) but when I do, I still have a couple of issues:
1. AttributeError mentioning a missing custom. type. This may have been solved by running `load_enums` first which leads to #2
2. AttributeError mentioning `ua.Bit`. These two pages stuck out to me because they are defined in the protocol:
* https://reference.opcfoundation.org/v104/Core/docs/Part5/E.5/
* https://reference.opcfoundation.org/v104/Core/docs/Part5/E.6/

What I believe is happening at this point is we're not able to load types like `ua.Bit` in `#2` in our python library for this older schema. So I'm trying to figure out how to load these type dictionaries. 
* `import_xml` seems like it's expecting to load a different type of schema for nodes so that doesn't work  
* There is a OPCTypeDictionary builder but that doesn't actually load from a file so I might have to parse and build it myself.

deeTEEcee, 2022-01-27:
hopefully im on the right path :)

deeTEEcee, 2022-01-27:
it looks like we don't support parsing OpaqueTypes so i might have to implement that and understand more of the internals.

AndreasHeine, 2022-01-28:
did you mean: 
https://github.com/FreeOpcUa/opcua-asyncio/blob/bbb7155040263131e47dce7456b6420f8264c663/asyncua/client/client.py#L670

oroulet, 2022-01-28:
Make s minimale example code calling load_data_type_definition() and show us the output and the error of you get one

deeTEEcee, 2022-01-28:
@AndreasHeine no, i specifically meant `load_type_definitions`, not `load_data_type_definitions` because of https://github.com/FreeOpcUa/opcua-asyncio/issues/506#issuecomment-808902331

closed 2023-03-29 by schroeder-