FreeOpcUa/opcua-asyncio #785: ItemSubscriptionModified callback is not triggered on server-callback example

saramonteiro, 2022-02-03:
**Describe the bug** <br />
The server-callback example is not triggering the callback when the attribute is changed by the client.
The creation and deletion callback is working.

**To Reproduce**<br />
1. Run the `server-callback` example.
2. Open a client session and change the `MyVariable` variable.

**Expected behavior**<br />
I expected to see `modify_monitored_items` running.

**Version**<br />
Python-Version:<br /> 3.10.1
opcua-asyncio Version: Master branch > last commit hash:  8631c0a7ecd920d0716b2965c2f3ad040405f090.

**Workaround**<br />
To catch the nodes' data changes, for now I used the approach from `client-subcription`.

swamper123, 2022-02-03:
"modify_monitored_items" means, that you changed e.g. the sampling interval. It's about changes _how_ you subscribe sth.. 
So not modifing the variable value, but the monitored item, which is about the Node.
If you want to get events of value changing, you have to subscribe the Node.

saramonteiro, 2022-02-03:
> "modify_monitored_items" means, that you changed e.g. the sampling interval. It's about changes _how_ you subscribe sth.. So not modifing the variable value, but the monitored item, which is about the Node. If you want to get events of value changing, you have to subscribe the Node.

Thanks for the clarification!

closed 2022-02-03 by saramonteiro