FreeOpcUa/opcua-asyncio #808: OPC-UA error (Bad Too Many Sessions)

heritopo, 2022-02-16:
Hello,
I am beginner about this topic.
I am trying to get multiple data from client (front cover station by FESTO) using python code. when i run the code, there was an error "Bad Too Many Session" as below picture.
[image]

server address : opc.tcp://10.48.134.10:4840
Signal address
[image]
[image]
picture of front cover station from festo 

Did the error appear due to code or server ?
Would you like to give solution regarding this ?

Thank you

AndreasHeine, 2022-02-16:
BadToManySessions means that the OPC UA Server has no more resources for another Session!
Most Embedded small OPC UA Devices have just a few available, you need to check the manual for that device sometimes you can crank that session count up a little.

heritopo, 2022-02-16:
Thank you for your explanation @AndreasHeine
I will check the manual and do it as your sugestion

AndreasHeine, 2022-02-17:
@heritopo ?

heritopo, 2022-02-17:
sorry @AndreasHeine, 
i can't find the literature of OPC UA server in the module,
do you have reference ?
and i realize that i should not make discussion in here. it supposed to be in discussion session.
if possible, I will transfer my issue in the discussion session.
Is it ok ?
thanks