FreeOpcUa/opcua-asyncio #810: AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'

BenediktEnger, 2022-02-17:
I'm having trouble reading an ExtensionObject from an Server running on <1.04. The Server runs on a B&R PLC.
When calling load_type_definition() I get folllowing error:

```
Deprecated since spec 1.04, call load_data_type_definitions

@dataclass
class FetchResultDataType:

    
    FetchResultDataType structure autogenerated from xml
    

@dataclass
class TransferResultErrorDataType:

    
    TransferResultErrorDataType structure autogenerated from xml
    

    Status:ua.Int32 = 0
    Diagnostics:ua.DiagnosticInfo = ua.DiagnosticInfo()

@dataclass
class TransferResultDataDataType:

    
    TransferResultDataDataType structure autogenerated from xml
    

    SequenceNumber:ua.Int32 = 0
    EndOfResults:ua.Boolean = True
    ParameterDefs:List[ua.ParameterResultDataType] = field(default_factory=list)

Failed to execute auto-generated code from UA datatype: 

@dataclass
class TransferResultDataDataType:

    
    TransferResultDataDataType structure autogenerated from xml
    

    SequenceNumber:ua.Int32 = 0
    EndOfResults:ua.Boolean = True
    ParameterDefs:List[ua.ParameterResultDataType] = field(default_factory=list)
Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\structures.py", line 281, in _generate_python_class
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 12, in TransferResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
```

Reading this issue https://github.com/FreeOpcUa/opcua-asyncio/issues/509 it seems like the error has been fixed but I still get it. 
I'm running the current master (0.9.92)

jrast, 2022-04-13:
This specific issue is caused by the type annotation which references a type which is not yet defined in the `ua` namespace. So to fix this, the code generator of the `Struct`

https://github.com/FreeOpcUa/opcua-asyncio/blob/cfe23db170473a0a8e28665c9f24c80b34fbde30/asyncua/common/structures.py#L74

can be adjusted such that the type annotation uses quotion marks:
https://github.com/FreeOpcUa/opcua-asyncio/blob/cfe23db170473a0a8e28665c9f24c80b34fbde30/asyncua/common/structures.py#L88
should be updated to:
````python
uatype = f"List['{uatype}']"
````

**However** this does not completly solve the problem as the order in which all the structs are created and assigned to the `ua` namespace is important if structures are nested and a actuall instance of a structure is created. For example the following code cannot be evaluated as some of the types are not yet defined:
````python
@dataclass
class Sample_System_type:

    '''
    Sample_System_type structure autogenerated from xml
    '''

    Assembly:ua.Sample_SystemDataAssemblyType = ua.Sample_SystemDataAssemblyType()  # --> This will fail!
    State:ua.Sample_SystemState_enum = ua.Sample_SystemState_enum()
````
There need to be a more elaborated way to resolve the dependencies between the types...

jrast, 2022-04-26:
I opened a pull request regarding this issue. Please have look at it and provide your inputs.

closed 2023-03-29 by schroeder-