FreeOpcUa/opcua-asyncio #813: Node parsing breaks when Browse Name contains colon

Randelung, 2022-02-18:
I have a call
```python
await node.add_object(f'ns={self.__namespace};g={guid}', name)
```
where `name` contains a colon ':' without designating a namespace.
I couldn't find any info towards the exclusion of colons in browse names - if I'm wrong this whole ticket is obviously void.

The creation of new objects (and likely other entities) assumes that the browse name argument contains a representation of a qualified name instead of a browse name. (`create_object` and others have `bname` parameters, after all, not `qname` or similar. Even the docstring says `browsename`.)

The parsing of said browse name into a qualified name should therefore be more lenient and not fail if the given name contains a colon without namespace identifier. I suggest to fall back to the general case if parsing fails. While that could be done [here](https://github.com/FreeOpcUa/opcua-asyncio/blob/033427b6a5bdf1a4a512e4976a410b6cc6291406/asyncua/common/manage_nodes.py#L29-L30), I'd go as far as to change the actual [`from_string` method on `QualifiedName`](https://github.com/FreeOpcUa/opcua-asyncio/blob/033427b6a5bdf1a4a512e4976a410b6cc6291406/asyncua/ua/uatypes.py#L607-L618), since it allows the general case as well.

```python
    @staticmethod
    def from_string(string):
        try:
            idx, name = string.split(":", 1)
            idx = int(idx)
            return QualifiedName(Name=name, NamespaceIndex=idx)
        except (TypeError, ValueError):
            if ":" in string:
                logger.warning(f"Error parsing string {string} into namespace index and browse name, falling back to 0:{string}")
        return QualifiedName(Name=string, NamespaceIndex=0)
```

Performancewise it's a little better than the original, too. Checking for the colon first means you go through the whole string once and a second time for the split if one is found. With the direct try/except approach, only a single pass is used. If there's no colon, the assignment fails with a ValueError. If the parse fails, a second iteration is used to produce a useful log message.
I'm not even sure the message is needed at all since having a colon in the browse name is not actually an unexpected situation.

schroeder-, 2022-02-18:
You should't use namespace = 0 at all. Namespace zero is reserved for opc ua specification, even for browse names. So if you don't instance object with objecttypes form namespace 0, always add a namespace in the qualified name.

oroulet, 2022-02-18:
Create a pull request

Randelung, 2022-02-18:
Okay, I've been doing a little more digging using the OPC UA .NET reference implementation and the OPC UA spec.

Alexander, you're obviously right. Up until today I was unaware that there's a namespace for browse names, too, and would have happily continued to abuse namespace 0, had there not been new nodes that contain a colon. (I worked mostly with the included server examples that don't show the correct use of nodeid/browsename combination.) I will ameliorate my code to explicitly use my own namespace, which will solve my issue (and it _is_ my issue).

I found that the namespace can be omitted (and defaults to 0) if there's no colon in the browse name (see e. g. [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/3dc1d4beba31ef5c0d0ade944ad688cd542b28fd/Stack/Opc.Ua.Core/Types/BuiltIn/QualifiedName.cs#L328-L340)). Based on the Alexander's comment I was exploring a path of enforcing an index prefix, but that would be wrong.
Additionally, that means the case where the browse name _does_ contain a colon, a namespace prefix _must_ be present, meaning the implementation is correct. It's my misuse of browse names that are actually qualified names that imply a namespace 0 which are the problem.

I would like to make a suggestion about `Node.add_object` and the like, though. As soon as  `_parse_nodeid_qname` is called in all of the methods, the argument formerly referenced to as `bname` is used as a qualified name, practically promoting a browse name that doesn't need a namespace to a qualified name that does (if only in spirit). `add_object` etc. either take a node ID and a qualified name, or a namespace index and a browse name that expands into `('ns={idx};i={autoinc}', '{idx}:bname')`. The current signature of `(self, nodeid, bname, ...)` is a mix of both. Consistent would be `(self, nodeid, qname, ...)` or `(self, ns_index, bname, ...)`. It's just a quality of life improvement to make it a little more intuitive.
This could be combined with docstrings for the `Node` methods to explain the `objecttype` and `instantiate_optional` in a QoL update.

(To go above and beyond, refactor the methods to use a dictionary instead of parameters, if not different method names. The ambiguity makes it hard to read. It even lead you to define `_parse_nodeid_qname` with `(*args)` instead of proper parameters because it's impossible to name lol. I'm not doing a PR for that, though, because I feel that exceeds my involvement on your library, and you're probably aware anyway.)

Cheers!

closed 2022-02-18 by Randelung