FreeOpcUa/opcua-asyncio #816: Client missing a way to access Event Object components (hasComponent reference)

sardware, 2022-02-22:
**Describe the bug** <br />
On an OPC UA client, I have to subscribe to an event EventTypeB which has not only properties/values but also Object nodes.

Types:
- Custom EventType A, subtype of BaseEventType, with 20 properties
- Custom EventType B is a subtype of EventTypeA and has also 3 Object components (each with their own variables and sub components)

I don't know how to access these Object components because with the current Event class I can access only the properties.

**To Reproduce**<br />
```
class SubHandler:
    async def event_notification(self, event):
        print(event.get_event_props_as_fields_dict()) # it only prints the dict with the 20 properties
        # logger.info(await event.get_children()) # Not working: Event is not a node
        logger.info(type(event))
        await asyncio.sleep(0.1)

if __name__ == '__main__':
    async with Client(url=opcua_url) as client:
        base, base_event = await find_base_object(client)
        await client.load_data_type_definitions() # I don't know if it is useful
        my_event = await search_child(base_event, 'MyEventTypeB')
        my_event_type = await get_filter_from_event_type([my_event])
        sub_handler = SubHandler()
        sub = await client.create_subscription(100, sub_handler)
        handler = await sub.subscribe_events(base, evfilter=my_event_type)
        while True:
            await asyncio.sleep(0.1)

```
This code only prints the 20 properties of EventTypeA and <class 'asyncua.common.events.Event'>

**Expected behavior**<br />
I would like to find a way, even ugly, to access the event's Object subnodes

**Version**<br />
Python-Version: 3.8.12<br />
opcua-asyncio Version (e.g. master branch, 0.9): 0.9.92

Thank you

KUZJurk, 2022-02-23:
In our local repository [KUZ-Leipzig/opcua-asyncio](https://github.com/KUZ-Leipzig/opcua-asyncio) I have implemented a way so that properties and variables of a EventType's objects are getting monitored too. It can probably help you with your issue.  
I would appreciate your feedback.

schroeder-, 2022-02-23:
You can construct your own ContentFilter to only get the event values you want. 
I add an example where you get EventType, Message, ActiveState/Id and ConditionName of an AlarmConditionType.
```
content_filter = ContentFilter(
            [
                ContentFilterElement(
                    FilterOperator.OfType,
                    [LiteralOperand(Variant(NodeId(ObjectIds.AlarmConditionType)))],
                )
            ]
        )
        filt = EventFilter(
            [
                SimpleAttributeOperand(
                    BrowsePath=[QualifiedName("EventType")],
                    AttributeId=AttributeIds.Value,
                    TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
                ),
                SimpleAttributeOperand(
                    BrowsePath=[QualifiedName("Message")],
                    AttributeId=AttributeIds.Value,
                    TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
                ),
                SimpleAttributeOperand(
                    BrowsePath=[QualifiedName("ActiveState"), QualifiedName("Id")],
                    AttributeId=AttributeIds.Value,
                    TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
                ),
                SimpleAttributeOperand(
                    BrowsePath=[QualifiedName("ConditionName")],
                    AttributeId=AttributeIds.Value,
                    TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
                ),
            ],
            WhereClause=content_filter,
        )
```

sardware, 2022-02-24:
> In our local repository [KUZ-Leipzig/opcua-asyncio](https://github.com/KUZ-Leipzig/opcua-asyncio) I have implemented a way so that properties and variables of a EventType's objects are getting monitored too. It can probably help you with your issue. I would appreciate your feedback.

That is great, thank you, I will test it on the next week and give you a feedback! 

> You can construct your own ContentFilter to only get the event values you want. I add an example where you get EventType, Message, ActiveState/Id and ConditionName of an AlarmConditionType.

Good to know, unfortunately I just started studying Alarms and Events, therefore more documentation would be appreciated :)

sardware, 2022-03-03:
> In our local repository [KUZ-Leipzig/opcua-asyncio](https://github.com/KUZ-Leipzig/opcua-asyncio) I have implemented a way so that properties and variables of a EventType's objects are getting monitored too. It can probably help you with your issue. I would appreciate your feedback.

We did our test with your version, unfortunately the OPC UA server hangs and crashes as soon as we subscribe to this "heavy" event. We have to restart the machine every time.
With the OPC UA sync library, we can receive it correctly and get its object sub-nodes.

schroeder-, 2022-03-03:
Can you post the result of this print statement?
`print(repr(my_event_type ))`
I think it's most likely that either the server or client logic fails to provide a good eventfilter.

sardware, 2022-03-03:
> Can you post the result of this print statement? `print(repr(my_event_type ))` I think it's most likely that either the server or client logic fails to provide a good eventfilter.

Thank you, here it is:

> EventFilter(SelectClauses=[SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='AverageCycleTime')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxBadPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxCycleCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxGoodPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxId')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='BoxTestSamplesCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='CavityCycleQuality')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='CurrentLotName')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='CycleQuality')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='CycleTime')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobBadPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobCycleCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobGoodPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobName')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobPartsCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobStatus')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='JobTestSamplesCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=20, Name='MachineCycleCounter')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', Namesp [... cut ...] >, FilterOperands=[SimpleAttributeOperand(TypeDefinitionId=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), BrowsePath=[QualifiedName(NamespaceIndex=0, Name='EventType')], AttributeId=<AttributeIds.Value: 13>, IndexRange=None), LiteralOperand(Value=Variant(Value=NodeId(Identifier='WiBaCycleParametersEventType', NamespaceIndex=18, NodeIdType=<NodeIdType.String: 3>), VariantType=<VariantType.NodeId: 17>, Dimensions=None, is_array=False))])]))

KUZJurk, 2022-03-03:
The identifier attribute of the NodeIds shouldn't be a string, doesn't it?