FreeOpcUa/opcua-asyncio #817: Unable to retrieve AutoId DataTypes from server via load_data_type_definitions

chrizzlybeard, 2022-02-22:
When trying to access the DataTypes, I've been following the example from https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-custom-structures-and-enums.py, but with a sync client. I'm not too familiar with OPC UA, I hope my description is sufficient (and the problem is not me being an idiot somehow).

When I run the server from https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-custom-structures-amd-enums.py, the modified client example works: 

`def main():
    with Client(url="opc.tcp://localhost:4840") as client:
        client.load_data_type_definitions()
        print(ua.MyStruct)

if __name__ == "__main__":
    main()
`

`
> <class 'MyStruct'>
`

Accessing the server via UaExpert, the Address Space looks like this:

[image]

When I try to run the same on my actual server (it's a camera system with an integrated OPC UA server), I get the following logging info, indicating that the server actually finds a bunch of DataType nodes:

`
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=334, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) ComplianceLevel OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=334, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=0, Name='ComplianceLevel')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=398, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) EnumeratedTestType OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=398, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=0, Name='EnumeratedTestType')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=6244, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) DeviceHealthEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=6244, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=2, Name='DeviceHealthEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3013, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) AutoIdOperationStatusEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3013, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='AutoIdOperationStatusEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3003, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) DeviceStatusEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3003, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='DeviceStatusEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3009, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) LocationTypeEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3009, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='LocationTypeEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3016, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) RfidLockOperationEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3016, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='RfidLockOperationEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3015, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) RfidLockRegionEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3015, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='RfidLockRegionEnumeration')
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=3014, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) RfidPasswordTypeEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3014, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='RfidPasswordTypeEnumeration')
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.client.ua_client.UaClient:browse
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=341, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=0, Name='SoftwareCertificate')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=335, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=0, Name='SupportedProfile')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=6522, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=2, Name='FetchResultDataType')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=6525, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=2, Name='ParameterResultDataType')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3011, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='AntennaNameIdPair')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3023, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='DhcpGeoConfCoordinate')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3019, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='LocalCoordinate')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3004, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='Position')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3006, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='RfidSighting')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3029, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='Rotation')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3024, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='ScanDataEpc')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3001, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='ScanResult')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3010, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='ScanSettings')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3027, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=3, Name='WGS84Coordinate')
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=3002, NamespaceIndex=4, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=4, Name='LastAccessResult')
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
`

For reference, this is what the DataTypes look like in UaExpert:

[image]

Some are present (e.g. ScanResult), but some are missing (e.g. ScanData, the DataType I'm most interested in). Even the DataTypes that are present in the logging info can't be instantiated from the ua namespace:

`
with Client(ENDPOINT) as client:
    client.load_data_type_definitions()
    print(ua.ScanResult)
`

`
> Traceback (most recent call last):
  File "C:\projects\camera_test_2.py", line 35, in <module>
    print(ua.ScanResult)
AttributeError: module 'asyncua.ua' has no attribute 'ScanResult'
`

Any help is much appreciated. Please let me know if there is some information missing.

Best regards

chrizzlybeard, 2022-02-23:
Thank you for the quick reply. I noticed that I actually had loglevel set to Info... sorry. When I set it to debugging, I can actually see that the problem most likely originates in https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/common/structures104.py#L344 and might be a server problem? Relevant part from the logs:

```python
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=6244, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) DeviceHealthEnumeration OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=6244, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=2, Name='DeviceHealthEnumeration')
[...]
DEBUG:asyncua.common.structures104:ns=2;i=6244 has no DataTypeDefinition attribute
```

ScanData doesn't appear at all in the log of load_data_type_definitions.

UaExpert does recognize ScanData and allows me to call the method that requires a ScanData object as parameter, so it seems to be able to understand the server.

For reference, this is how the call dialog looks like:

[image]

Digging around on the server, I can find ScanData in two places:

[image]

The DataTypeDefinition attribute for this node says BadAttributeInvalid:

[image]

The other place:

[image]

doesn't seem to have a DataTypeDefinition attribute at all:

[image]

Maybe this is also a problem with my understanding of OPC UA. How does UaExpert manage to call the method (it seems capable of understanding ScanData)?

Is the first ScanData node the actual TypeDefinition (then it would make sense it doesn't have a DataTypeDefinition itself)?

This is probably off topic, but is there a way for the client to build custom DataTypes/Structs in a way they can be instantiated from Python and passed into a method? That would be a second avenue for me to solve my underlying problem.

Thanks again for your time!

schroeder-, 2022-02-23:
I think we don't support any datatypes which are subtype of union, because on the wire they are encoded/decoded different, then normal structs(datatypes).

schroeder-, 2022-02-23:
You can use the ExtensionObject direct and encode/decode as [union](https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.8/). But this is realy advanced stuff.

rmshs, 2022-02-24:
Hi, 

I happen to be facing the same issue with a datatype that is _**however**_ a subtype of String. It is defined here: [CodeTypeDataType](https://reference.opcfoundation.org/v104/AutoID/v101/docs/9.1.3/)

For me load_data_type_definitions also is not able to create the datatype on the fly and returns:

`DEBUG:asyncua.common.structures104:ns=3;i=3031 has no DataTypeDefinition attribute`

When tested with UaExpert it does not have issues handling the datatype, eventhough UaExpert can **_also_** not access the DataTypeDefinition:
[image]

Any hints on how I could build this datatype in python would be appreciated. 

Kind regards

oroulet, 2022-02-24:
Does none of your nodes have  DataTypeDefinition or are only some structs missing them?
Anyway you can try calling load_enums() and load_type_defitnioins() intead of load_data_type_definitions() which are the older version of protocol. Unfortunately the python implementation if not complete.

concerning Union I never looked at how to implement it because I had no use for it and no example, but i cannot imagine it is hard

schroeder-, 2022-02-26:
In #825   I added support for unions via DatatypeDefinition. Also you can now manually create union class.
For your case this would be:

```
@dataclass
class ScanData(ua.UaUnion):
    data_type = ua.NodeId.from_string("ns=3;i=6131")

    Encoding: ua.Byte = field(default=0, repr=False, init=False, compare=False)
    Value: Union[None, ua.ByteString, ua.String, ua.Epc, ua.Custom] = field(default=None, init=False)

    @property
    def ByteString(self) -> Optional[ua.String]:
        if self.Encoding == 1:
            return self.Value
        return None

    @ByteString.setter
    def ByteString(self, value: ua.ByteString) -> None:
        self.Value = value
        self.Encoding = 1

    @property
    def String(self) -> Optional[ua.String]:
        if self.Encoding == 2:
            return self.Value
        return None

    @String.setter
    def String(self, value: ua.String) -> None:
        self.Value = value
        self.Encoding = 2

    @property
    def Epc(self) -> Optional[ua.Epc]:
        if self.Encoding == 3:
            return self.Value
        return None

    @Epc.setter
    def Epc(self, value: ua.Epc) -> None:
        self.Value = value
        self.Encoding = 3

    @property
    def Custom(self) -> Optional[ua.Custom]:
        if self.Encoding == 4:
            return self.Value
        return None

    @Custom.setter
    def Custom(self, value: ua.Custom) -> None:
        self.Value = value
        self.Encoding = 4
```

chrizzlybeard, 2022-03-01:
Thanks again everyone for helping, especially @schroeder- for taking the time to implement union! From my side, the issue is resolved, so unless @rmshs has further questions I'd close the issue.

rmshs, 2022-03-01:
> Does none of your nodes have DataTypeDefinition or are only some structs missing them? Anyway you can try calling load_enums() and load_type_defitnioins() intead of load_data_type_definitions() which are the older version of protocol. Unfortunately the python implementation if not complete.

No, they do not, however I figured that using the parenttype (String in my case) is possible, as long as the input is restricted to the definition in [CodeTypeDataType](https://reference.opcfoundation.org/v104/AutoID/v101/docs/9.1.3/). Long story short, it works :)

@schroeder- Thanks for the Union implementation!
@chrizzlybeard That is all from my side too, sorry for hijacking your issue :)

closed 2022-03-01 by chrizzlybeard