FreeOpcUa/opcua-asyncio #827: Can not connect to kepserver using client_set_security_string

NieYanzhai, 2022-03-02:
### kepserver opcua config:
[image]

### python code:
[image]

### error message:
C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\Scripts\python.exe C:/Users/yanzh/Desktop/codes/connect_to_opcua_security/main.py
Exception raised while parsing message from server
Traceback (most recent call last):
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\ua_client.py", line 78, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\common\connection.py", line 349, in receive_from_header_and_body
    chunk = MessageChunk.from_header_and_body(self.security_policy, header, body, use_prev_key=False)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\common\connection.py", line 60, in from_header_and_body
    decrypted = crypto.decrypt(data.read(len(data)))
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\crypto\security_policies.py", line 198, in decrypt
    return self.Decryptor.decrypt(data)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\crypto\security_policies.py", line 309, in decrypt
    decrypted += self.decryptor(self.client_pk,
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\crypto\uacrypto.py", line 123, in decrypt_rsa_oaep
    text = private_key.decrypt(
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\cryptography\hazmat\backends\openssl\rsa.py", line 433, in decrypt
    return _enc_dec_rsa(self._backend, self, ciphertext, padding)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\cryptography\hazmat\backends\openssl\rsa.py", line 87, in _enc_dec_rsa
    return _enc_dec_rsa_pkey_ctx(backend, key, data, padding_enum, padding)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\cryptography\hazmat\backends\openssl\rsa.py", line 147, in _enc_dec_rsa_pkey_ctx
    res = crypt(pkey_ctx, buf, outlen, data, len(data))
TypeError: initializer for ctype 'unsigned char *' must be a cdata pointer, not bytearray
disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "C:\Program Files\Python310\lib\asyncio\tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\main.py", line 20, in <module>
    asyncio.run(main())
  File "C:\Program Files\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\Python310\lib\asyncio\base_events.py", line 641, in run_until_complete
    return future.result()
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\main.py", line 14, in main
    async with client:
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\client.py", line 68, in __aenter__
    await self.connect()
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\client.py", line 246, in connect
    await self.open_secure_channel()
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\client.py", line 297, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\ua_client.py", line 280, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "C:\Users\yanzh\Desktop\codes\connect_to_opcua_security\venv\lib\site-packages\asyncua\client\ua_client.py", line 213, in open_secure_channel
    await asyncio.wait_for(self._send_request(request, message_type=ua.MessageType.SecureOpen), self.timeout)
  File "C:\Program Files\Python310\lib\asyncio\tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

Process finished with exit code 1

AndreasHeine, 2022-03-02:
@schroeder- ".pem" should works as well!

AndreasHeine, 2022-03-02:
@NieYanzhai did you add the client cert to the trusted list?

[image]

NieYanzhai, 2022-03-03:
> @NieYanzhai did you add the client cert to the trusted list?
> 
> [image]

### Yes, client cert added to the trust list.
[image]

### And,  I connected to the kepserver using opcua, but can't connect using ayncua.
[image]

NieYanzhai, 2022-03-03:
> @schroeder- ".pem" should works as well!

pem works.

jdess, 2022-03-24:
I'm facing the same issue with the exception that gets raised in the log of https://github.com/FreeOpcUa/opcua-asyncio/issues/827#issue-1157287320:
`TypeError: initializer for ctype 'unsigned char *' must be a cdata pointer, not bytearray`

On my side, I was able to work around this by going back to Python version 3.7.9. With version 3.10.1, I'm seeing the`TypeError` exception.

jdess, 2022-03-24:
After reviewing the code again, I assume that the better place to fix that is in `asyncua.client.ua_client.data_received`. This is where the data provided from `asyncio` has changed from `bytes` to `bytearray`. In fact, the type hint on this function is not correct any more, either:
```python
    def data_received(self, data: bytes):
        if self.receive_buffer:
            data = self.receive_buffer + data
            self.receive_buffer = None
        self._process_received_data(data)
```

In this function, if `data` is a `bytearray` and not `bytes`, we probably need to convert it from `bytearray` to `bytes` as shown in my previous post (https://github.com/FreeOpcUa/opcua-asyncio/issues/827#issuecomment-1077491842). This would make sure that we re-establish the legacy behavior where `data` was delivered as `bytes`. Something along the lines of:
```python
    def data_received(self, data: bytes or bytearray):
        # `data`` might be a `bytearray` in more recent Python versions
        # (see https://github.com/python/cpython/commit/568fb0ff4aa641329261cdde20795b0aa9278175)
        # However, the downstream code still expects to see `bytes`:
        if isinstance(data, bytearray):
            data = bytes(data)

        if self.receive_buffer:
            data = self.receive_buffer + data
            self.receive_buffer = None
        self._process_received_data(data)
```

closed 2023-03-29 by schroeder-