FreeOpcUa/opcua-asyncio #837: Impossible to import default schema `Opc.Ua.PLCopen.NodeSet2_V1.02.xml`

danmoser, 2022-03-08:
`asyncua.Server.import_xml()` fails when importing default schema `Opc.Ua.PLCopen.NodeSet2_V1.02.xml`.

PLCopen requires `Opc.Ua.Di.NodeSet2.xml` to be imported first since it is a dependency. Then, the following error is generated when importing PCLopen:

```
asyncua.common.xmlimporter:Importing xml node (QualifiedName(NamespaceIndex=3, Name='<TaskName>'), NodeId(Identifier=1008, NamespaceIndex=3, NodeIdType=<NodeIdType.Numeric: 2>)) as (QualifiedName(NamespaceIndex=3, Name='<TaskName>') NodeId(Identifier=1008, NamespaceIndex=3, NodeIdType=<NodeIdType.Numeric: 2>))
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=1008, NamespaceIndex=3, NodeIdType=<NodeIdType.Numeric: 2>))
Traceback (most recent call last):
  File "/home/user/acu_server.py", line 53, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/user/acu_server.py", line 40, in main
    async with HelloServer(
  File "/home/user/acu_server.py", line 30, in __aenter__
    await self.init()
  File "/home/user/acu_server.py", line 21, in init
    await self.server.import_xml(self.antMgr_filepath)
  File "/home/user/.local/lib/python3.9/site-packages/asyncua/server/server.py", line 585, in import_xml
    return await importer.import_xml(path, xmlstring)
  File "/home/user/.local/lib/python3.9/site-packages/asyncua/common/xmlimporter.py", line 110, in import_xml
    node = await self._add_node_data(nodedata, no_namespace_migration=True)
  File "/home/user/.local/lib/python3.9/site-packages/asyncua/common/xmlimporter.py", line 184, in _add_node_data
    node = await self.add_object(nodedata, no_namespace_migration)
  File "/home/user/.local/lib/python3.9/site-packages/asyncua/common/xmlimporter.py", line 305, in add_object
    res[0].StatusCode.check()
  File "/home/user/.local/lib/python3.9/site-packages/asyncua/ua/uatypes.py", line 318, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
```

`asyncua` version 0.9.92.

schroeder-, 2022-03-11:
I looked into it.  From the specs side the Nodeset is illegal, because it is missing ParentNodeId for <TaskName>. 
I tried to load it with a [Nodesetviewer ](https://github.com/digitaltwinconsortium/UANodesetWebViewer) und the node is missing there to. 
Maybe we should just log and discard the error.

GoetzGoerisch, 2022-03-11:
@schroeder- and @danmoser  please log a Mantis issue for the companion spec editors: https://github.com/OPCFoundation/UA-Nodeset#error-reporting

Direct link: https://mantis.opcfoundation.org/main_page.php

danmoser, 2022-03-11:
Thank you for the feedback. Bug report in Mantis https://mantis.opcfoundation.org/view.php?id=7858

closed 2022-03-11 by danmoser