FreeOpcUa/opcua-asyncio #841: Messages size is hardcoded and should be coherent in HEL/ACK and session

vmo6trl, 2022-03-16:
Hello,
I encountered an issue by using a freeopcua server with another client and trying to be compliant with the messages size supported.
In TCP UA Acknowledge message a fixed size of chunk is provided but no limit for the number of chunks or the total message size whereas in CreateSessionResponse message the MaxRequestMessageSize indicated is the size of a chunk. 
Therefore it becomes impossible to send several chunks to the server if the client checks and complies for the limit provided at session level.
After checking the library code I found the origin of the incoherence.
In `ua/uaprotocol_hand.py` we can find:
```python
class Acknowledge:
    ProtocolVersion: uatypes.UInt32 = 0
    ReceiveBufferSize: uatypes.UInt32 = 65536
    SendBufferSize: uatypes.UInt32 = 65536
    MaxMessageSize: uatypes.UInt32 = 0  # No limits
    MaxChunkCount: uatypes.UInt32 = 0  # No limits
```
whereas in `server/internal_session.py` we can find:
```python
async def create_session(self, params, sockname=None):
        self.logger.info('Create session request')
        result = ua.CreateSessionResult()
        result.SessionId = self.session_id
        result.AuthenticationToken = self.auth_token
        result.RevisedSessionTimeout = params.RequestedSessionTimeout
        result.MaxRequestMessageSize = 65536
```
I think `result.MaxRequestMessageSize ` might be set to 0 in order to be coherent with what is declared in Acknowledge message and allows the use of multi-chunks messages which are actually supported by the server. Moreover I guess those constant values might be configurable by application.

Best regards.

oroulet, 2022-03-16:
yes. Send a PR

schroeder-, 2023-03-29:
Fixed

closed 2023-03-29 by schroeder-