FreeOpcUa/opcua-asyncio #846: Importing NodeSet does not register the namespace to the server.nodes.namespaces shortcut

Derbster, 2022-03-25:
When importing an XML in a server, the newly created namespace is added to the namespace_array of the server but not to the server.nodes.namespaces shortcuts child nodes. 
This leads to a bug when importing models, which require one another, since the check for required models on the server is depending on the server.nodes.namespaces shortcut.
Importing [Opc.Ua.Di.NodeSet2.xml](http://opcfoundation.org/UA/DI/) works fine. Trying to afterwards import [Opc.Ua.Plc.NodeSet2.xml ](http://PLCopen.org/OpcUa/IEC61131-3/) leads to an exception, since the check for required models does not recognize, that [Opc.Ua.Di.NodeSet2.xml](http://opcfoundation.org/UA/DI/) is already imported. Commenting out the exception leads to the warning log, however the import passes and the server can be started.

apoigne, 2022-07-08:
I do have a similar problem: if in one XML file several namespaces are specified where e.g. the model of one namespace depends on a model of a earlier specified namespace , the only namespace in server.nodes.namespaces is the 'http://opcfoundation.org/UA'. Consequently, the method _check_required_models fails. since it depends on self.server.nodes.namespaces.get_children().

schroeder-, 2022-07-17:
The error is in the nodeset, correct nodesets should always add an object to self.server.nodes, which contains the these values.

apoigne, 2022-07-17:
Does this imply that SIOME generates incorrect XMLs?

schroeder-, 2022-07-17:
Don't think they are invalid. The just missing information. All nodesets i know off, add these informations. Maybe we could add the informations when they are missing, because they are in other places in the xml?

apoigne, 2022-07-17:
What are these informations?

A second observation: If you look for the children of server.nodes.namespaces directly after the server.init() you get

NamespaceUri = http://opcfoundation.org/UA/
 NamespaceVersion = 1.04.7
NamespacePublicationDate = 2020-11-20 00:00:00+00:00

which pretty old. So if you have a newer version http://opcfoundation.org/UA/ in your XML, you are done. Tried to find where the data are defined but failed. Should probably be update to the newest version.

schroeder-, 2022-07-17:
then you are using a old version, master should use 1.04.10 from 2021.
Added a check if the objects in namespaces exists, otherwise add them to the addressspace.

apoigne, 2022-07-18:
Version is 0.9.94. That is what I get from pip install  asyncua .

closed 2023-03-29 by schroeder-