FreeOpcUa/opcua-asyncio #849: Close Session Manually

Charlyx12, 2022-03-29:
Hi! How could I close the session of an active user manually from the server? For example, if the password of the client has expired I want the server to eject him. What sequence should I follow to stop the communication with only that client?

Thank you

oroulet, 2022-04-01:
Good question. Never thought of it and one probably need to make a PR

oroulet, 2024-11-25:
honestly @AiyionPrime I have not looked at that code for a long time but from what you are saying it sounds like of those many timeout sockets which are hard to handle correctly.

also I jsut did a small test now and calling stop() do end all client connection immediatly on my pc now.... so it must be a special case somehow

AiyionPrime, 2024-11-25:
Maybe a weird edge case between asyncua and open62541. Thanks for having looked into it though.
I'm unsure whether i pursue this further; for now it only affects tests code.

Once that changes, I'll come back for more :)