FreeOpcUa/opcua-asyncio #851: `new_enum` triggers an Error message

danmoser, 2022-03-29:
**Describe the bug** <br />
`asyncua.common.structures104.new_enum` triggers a **message** error. Functionality appears to be fine.

**To Reproduce**<br />
Any (successful) call of `new_enum` (eg., *opcua-asyncio/examples/server-custom-structures-amd-enums.py*)

**Expected behavior**<br />
We should not see an error message.

**Screenshots**<br />

```
INFO:asyncua.common.structures104:Registring Enum NodeId(Identifier=5, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) MyEnum OptionSet=False
INFO:asyncua.common.structures104:Registering data type NodeId(Identifier=5, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) QualifiedName(NamespaceIndex=2, Name='MyEnum')
ERROR:asyncua.common.structures104:tutu - EnumField(Value=2, DisplayName=LocalizedText(Locale=None, Text='tutu'), Description=LocalizedText(Locale=None, Text=None), Name='tutu') False 

class MyEnum(IntEnum):

    '''
    MyEnum EnumInt autogenerated from EnumDefinition
    '''

    titi = 0
    toto = 1
    tutu = 2

INFO:asyncua.ua.uatypes:registring new enum: MyEnum NodeId(Identifier=5, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>) <enum 'MyEnum'>
INFO:asyncua.common.structures104:Registering data type NumericNodeId(Identifier=258, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) QualifiedName(NamespaceIndex=0, Name='Node')
INFO:asyncua.common.structures104:Registering data type NumericNodeId(Identifier=285, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) QualifiedName(NamespaceIndex=0, Name='ReferenceNode')
```

**Version**<br />
Python-Version: 3.9 <br />
opcua-asyncio Version: 0.9.92

schroeder-, 2022-03-30:
This was some left over debug output, this is fixed in the current master.

danmoser, 2022-03-30:
Thanks @schroeder- , but I cloned the master today (30/Mar/2022) and the displayed message is still the same.

PR #863 (2022-04-27): remove useless print
Fixes #851
closed 2022-04-27 by oroulet