FreeOpcUa/opcua-asyncio #856: Client should be able to read Tag with a Bad Quality

Falital, 2022-04-07:
Hi everyone thanks a lot for your good job and this nice python library, i know this should be a discussion but i sadly am not able to start one the start button stays light green in my case.

We decide to use the library to automate some system test we currently run manually, a test case we have that works with another opcua client dictates that a tag is writen with a bad quality and then is read back to verify this works sadly this throws the following error
```
./env/lib/python3.8/site-packages/asyncua/ua/uatypes.py", line 318, in check
    raise UaStatusCodeError(self.value)
```
I checked the master code since i am using 0.9.92 and this check and throw is also there but in line 303.
Then I took a look at https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/ua/uatypes.py and found this  cryptic message in the check function that causes this 'Use the is_good() method if you do not want an exception.' and to be honest i dont see a way this could work as.

For me it seems like a without reason restriction and i have some ideas on how this restriction could be removed i would be happy to provide a pull request if you agree with me.

Sample code to reproduce this.

```
from asyncua.sync import Client
from asyncua import ua

client = Client('url')
try:
    client.connect()
    value = client.get_node('ns=1;i=9')
    variant= ua.Variant("_", ua.VariantType.Boolean)
    dataValue = ua.DataValue(variant)
    object.__setattr__(datavalue, 'StatusCode_', ua.StatusCode(ua.StatusCodes.Bad))
    value .set_value(dataValue)
    result = value .read_attribute(ua.AttributeIds.Value)
    print(result)
finally:
    client.disconnect()
```

oroulet, 2022-04-07:
use read_data_value() and you will get the entire DataValue without any check

Falital, 2022-04-07:
read_data_value internaly calls read_attribute which leads to the same issue, plus the sync wrapper does not have read_data_value i could add this as well on my pull request.

oroulet, 2022-04-07:
right. there is a check too. checking should be an option onread_attribute and it should be disabled for read_data_value I think

Falital, 2022-04-07:
I agree i will create a pull request to fix this.

PR #857 (2022-04-27): fix: read_attribute can optional not raise an exception
* Previously a client could not read values with bad status.
* This functionality can be toggled on and off.
* Added tests.

Fixes #856
commit ab06cdbf (2022-04-27): fix: read_attribute can optional not raise an exception

* Previously a client could not read values with bad status.
* This functionality can be toggled on and off.
* Added tests.

Fixes #856
closed 2022-04-27 by oroulet