FreeOpcUa/opcua-asyncio #858: UASocketProtocol Exception: Exception raised while parsing message from server

jrast, 2022-04-11:
**Describe the bug** <br />
I get the error shown in the traceback below from time to time.

**To Reproduce**<br />
Currently I don't know how to reproduce this error consistently. It just happens now and then.

**Traceback**<br />
````
ERROR:asyncua.client.ua_client.UASocketProtocol:Exception raised while parsing message from server
Traceback (most recent call last):
  File "D:\SVN\FLOW\Python\AcoposDriver\venv\lib\site-packages\asyncua\client\ua_client.py", line 88, in 
_process_received_data
    data = bytes(buf)
TypeError: __bytes__ returned non-bytes (type bytearray)
````

**Version**<br />
Python-Version: Python 3.10.4 on Windows 10<br />
opcua-asyncio Version: 0.9.92

AndreasHeine, 2022-04-11:
What OPC UA Server is it? Wireshark-Trace?

jrast, 2022-04-11:
A workaround, which seems to fix the issue so far, is the follwoing:

````python
# In ua_clinet.py

def _process_received_data(self, data: bytes):
    # OLD: buf = ua.utils.Buffer(data)
    buf = ua.utils.Buffer(bytes(data))
    # Rest of the function as is.
````

While the type annotation of _process_received_data for `data` is `bytes`, it seems like the input is actually of type `bytearray`?

jrast, 2022-04-11:
The issue arised with a B&R Server I can provide more information about the server tomorow, the same for a Wireshark Trace.

Maybe I can try to reproduce it with the python server, but as noted: It's not that easy to reproduce...

jrast, 2022-04-11:
I think it's an issue with the Buffer class:

````python
In [1]: from asyncua.common.utils import Buffer

In [2]: buf = Buffer(bytearray([1, 2, 3, 4, 5, 6]))

In [3]: buf
Out[3]: Buffer(size:6, data:bytearray(b'\x01\x02\x03\x04\x05\x06'))

In [4]: bytes(buf)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [4], in <cell line: 1>()
----> 1 bytes(buf)

TypeError: __bytes__ returned non-bytes (type bytearray)
````

jrast, 2022-04-11:
I was able to reproduce the error with the python server and intercept the communication with wireshark. I tried to remove as much as possible from the reproduction script and ended up with the following capture:
[WireShar-Capturek-Issue858.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/8467191/WireShar-Capturek-Issue858.zip)
[image]

I think it's frame 97 which causes the issue: This frame contains multiple "OpcUa Binary Protocol" messages. If this frame is processed by `UASocketProtocol._process_received_data` the method reaches the `data = bytes(buf)` line, which is not the case for frames with only a single message.

According to my understanding the bug can be fixed in different ways:
- Convert anything to bytes in `UASocketProtocol.data_received`. This way there are no changes required later on as this is the place where data is fed into the protocol processing. To my understanding the [official python documentation](https://docs.python.org/3/library/asyncio-protocol.html#asyncio.Protocol.data_received) about `data_received` is somewhat missleading with "data is a non-empty bytes object" when it's actually a bytearray object. So here would be a simple point to fix this and make the data actually `bytes`.
- Same as above, but fix it in `UASocketProtocol._process_received_data`.
- Fix the `ua.utils.Buffer.__bytes__` implementation to make sure that `bytes` are returned even if the internal buffer is `bytearray`

If one of these fixes is accepted I'm happy to provide the PR for this.

jrast, 2022-04-13:
Just as an update to this: The last two days I was using the following patch wich fixes the implementation of `Buffer.__bytes__`. Since then I never again had the error mentioned above.

````python
# Monkeypatch for asyncua bug 858
# See https://github.com/FreeOpcUa/opcua-asyncio/issues/858
def patch_bug858():
    from asyncua.common.utils import Buffer

    def __bytes__(self):
        return bytes(self._data[self._cur_pos :])

    Buffer.__bytes__ = __bytes__

patch_bug858()
del patch_bug858
````

jrast, 2022-04-22:
I just realised that the proposed fix was already commited to master in https://github.com/FreeOpcUa/opcua-asyncio/commit/9a692cad050829ef13b18cfb94fee95bc1e7c6c6. So with the next release this bug will be fixed.

closed 2022-04-22 by jrast