FreeOpcUa/opcua-asyncio #874: Unable to connect to new server using Aes128_Sha256_RsaOaep

jfrechard, 2022-05-03:
**Problem ** <br />

I am trying to connect to a new generation OPC UA server.
Connexion works fine with uaExpert with these parameters : 

[image]

However, using asyncua I obtain : 

`asyncua.ua.uaerrors._base.UaError: Unsupported security policy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep`

My code that work with another server : 
```
async def task():
    url = influx.machine_cfg.opc_url 
    client = Client(url=url)
    client.set_user("toto")#influx.machine_cfg.opc_user)
    client.set_password("toto")#influx.machine_cfg.opc_password)
    #print(url,influx.machine_cfg.opc_user,influx.machine_cfg.opc_password,opc.cert,opc.private_key)
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        certificate=opc.cert,
        private_key=opc.private_key
    )
    async with client:
        handler = opc.SubscriptionHandler()
        subscription = await client.create_subscription(100, handler)
        nodes = []
        for node in list_var_nodes:
            nodes.append(client.get_node(node))
        print(nodes)
        # We subscribe to data changes for two nodes (variables).
        await subscription.subscribe_data_change(nodes)
        # We let the subscription run for ten seconds
        await asyncio.sleep(3600000)
        # We delete the subscription (this un-subscribes from the data changes of the two variables).
        # This is optional since closing the connection will also delete all subscriptions.
        await subscription.delete()
        # After one second we exit the Client context manager - this will close the connection.
        await asyncio.sleep(1)
```

AndreasHeine, 2022-05-03:
you don't use the correct security policy!?

not sure what this is -> "SecurityPolicyBasic256Sha256" (DataType?)

jfrechard, 2022-05-03:
I use it like example shows : 

'from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256'

AndreasHeine, 2022-05-03:
how did you generate your cert?

jfrechard, 2022-05-03:
Using openSSL. 
My client works on another server (different hardware).

AndreasHeine, 2022-05-03:
weird!

the error happens here:
https://github.com/FreeOpcUa/opcua-asyncio/blob/259811d5ad11c47c8d2b2289a8a2783b97c79421/asyncua/crypto/security_policies.py#L705

but we dont support the "Aes128_Sha256_RsaOaep" policy only SecurityPolicyBasic256Sha256, SecurityPolicyBasic256, SecurityPolicyBasic128Rsa15

jfrechard, 2022-05-03:
Ok, 
I thought that the server support all policies and we can choose one of them.

If I remove user and password, it works ...

jfrechard, 2022-05-03:
Thanks a lot !

AndreasHeine, 2022-05-03:
:astonished:

jfrechard, 2022-05-03:
? 
why are you astonished? because of my thought ? because I said "thanks' ?

AndreasHeine, 2022-05-03:
> If I remove user and password, it works ...

!?

jfrechard, 2022-05-03:
yes for sure.
I have an access to a simple web interface to configure the OPCUA server of my device, if I remove user and password of this configuration and if I remove it in my code it works.

I change nothing except that in my code 
'''
    client.set_user("")
    client.set_password("")
'''

I think the manufacturer of the device said that if you let user and password blank in the server configuration, it will start the server without encryption...

AndreasHeine, 2022-05-03:
what server/vendor is it? (for my own curiosity ^^)

jfrechard, 2022-05-03:
Its an HMA box from Siemens.

Its a computer running windows 10 IoT, you plug it on an hold Siemens CNC and he will expose variables you choose using OPCUA.

Its a simple way to add an OPC UA server to an old CNC without OPC capability.

closed 2023-03-29 by schroeder-