FreeOpcUa/opcua-asyncio #875: `new_struct(..., is_union=True)` is not working as expected

danmoser, 2022-05-03:
**Describe the bug** <br />
I've created a custom struct with `asyncua.common.structures104.new_struct()` and tried to convert it to "Union". Then the `.read_value()` from the nodeset is empty. If disable keep it as a regular struct (ie., `is_union=False`) it works good.

**Screenshots**<br />
`is_union=False`:
[image]
`is_union=True`:
[image]

**Version**<br />
Python-Version: 3.10
opcua-asyncio Version: master branch, 0.9.92 from early May 2022

schroeder-, 2022-05-04:
Works as expected.
A Union is Null or **ONE** of its Fields. The default value in asyncua is Null. See: https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.8/

danmoser, 2022-05-16:
Thanks @schroeder- . I'm sorry for the false flag.

closed 2022-05-16 by danmoser