FreeOpcUa/opcua-asyncio #883: Not enough data left in buffer。

EvashuangshuangFeng, 2022-05-16:
Nested structure：

await client.load_type_definitions()
after = await struct.read_value()
---------------------------------------

File "C:\Program Files\Python38\lib\site-packages\asyncua\ua\ua_binary.py", line 73, in unpack
    b = _Bytes.unpack(data)
  File "C:\Program Files\Python38\lib\site-packages\asyncua\ua\ua_binary.py", line 61, in unpack
    return data.read(length)
  File "C:\Program Files\Python38\lib\site-packages\asyncua\common\utils.py", line 59, in read
    raise NotEnoughData(f"Not enough data left in buffer, request for {size}, we have {self._size}")
asyncua.common.utils.NotEnoughData: Not enough data left in buffer, request for 256, we have 95

EvashuangshuangFeng, 2022-05-17:
@schroeder-  
1. Read structure fails with string datatype.
2. When I delete the string datatype, read structure succeeds, but the value of structure and controller do not match.

2.1 This is value from asyncua
[image]

2.2  This is value from Uaexpert.
[image]

EvashuangshuangFeng, 2022-05-17:
> Looks like a we don't support some special cases in the encoding.
> 
> What server you are running against? Which version of asyncua are you using? (Is it possible to use current master?) Is it possible to get the datatype description of the datatypes in question or as nodeset.xml?

closed 2022-05-17 by EvashuangshuangFeng

oroulet, 2022-05-18:
I do not understand is that issue solved?