FreeOpcUa/opcua-asyncio #884: Server sync method create_subscription is missing.

dannmartens, 2022-05-16:
**Describe the bug** <br />
Server sync method create_subscription is missing, the code fails with:
```
AttributeError: 'Server' object has no attribute 'create_subscription'
```

**To Reproduce**<br />
Uncomment subscription in _opcua-asyncio/examples/sync/server-example.py_
```
            # enable following if you want to subscribe to nodes on server side
            #handler = SubHandler()
            #sub = server.create_subscription(500, handler)
            #handle = sub.subscribe_data_change(myvar)
```

**Expected behavior**<br />
A demonstration of a working subscription.

**Version**<br />
Python-Version: 3.8.10<br />
opcua-asyncio Version (e.g. master branch, 0.9): 0.9.92

dannmartens, 2022-05-16:
To put this in context, this issue cropped up after upgrading asyncua for an existing code base from 0.9.0 to 0.9.92.

With 0.9.0, an internal server subscription could be created synchronously by running an async task:
```
        # vars contains the targets
        handler = SubscriptionHandler()
        subscription = await self.server.aio_obj.create_subscription(100, handler)
        handles = await subscription.subscribe_data_change(vars)
```

With 0.9.92, the same code did not fail, but the subscription did not work anymore.

When I was rewriting the subscribing step with pure synchronous calls, I found out asyncua.sync.Server did not have _create_subscription_.

I fixed this by monkey patching:
```
from asyncua.sync import Client, Server
Server.create_subscription = Client.create_subscription
```

As it turns out, the Client implementation seems to do the job.

oroulet, 2022-05-18:
https://github.com/FreeOpcUa/opcua-asyncio/pull/887

So you see how to easily add that kind of function. If you tell me it works, I will merge

dannmartens, 2022-05-20:
It has been running in our staging environment for a couple of days, now. I'm planning additional tests, next Tuesday.

Please stand by, I will report my findings.

dannmartens, 2022-06-08:
It's been running more than two weeks, without any issues.

closed 2022-06-08 by dannmartens