FreeOpcUa/opcua-asyncio #885: Reconciliator, has undefined attributes/functions

vruge, 2022-05-16:
static code analyse with mypy complains in the following places

- [ ] `Reconciliator` has no attribute `hook_mi_request` 

https://github.com/FreeOpcUa/opcua-asyncio/blob/4f48eb2df8d998e8f576779fbae8c5c913fddab5/asyncua/client/ha/reconciliator.py#L323

https://github.com/FreeOpcUa/opcua-asyncio/blob/4f48eb2df8d998e8f576779fbae8c5c913fddab5/asyncua/client/ha/reconciliator.py#L293

- [ ] `Reconciliator` has no attribute  `hook_mi_request` 

https://github.com/FreeOpcUa/opcua-asyncio/blob/4f48eb2df8d998e8f576779fbae8c5c913fddab5/asyncua/client/ha/reconciliator.py#L384

- [ ] `Reconciliator` has no attribute  `hook_add_to_map`  

https://github.com/FreeOpcUa/opcua-asyncio/blob/4f48eb2df8d998e8f576779fbae8c5c913fddab5/asyncua/client/ha/reconciliator.py#L411

- [ ] `Reconciliator` has no attribute  `hook_del_from_map`

https://github.com/FreeOpcUa/opcua-asyncio/blob/4f48eb2df8d998e8f576779fbae8c5c913fddab5/asyncua/client/ha/reconciliator.py#L432

related to #878

schroeder-, 2022-05-17:
I don't know anything about the ha client. For me it looks like debug hooks that are left in the code?

ZuZuD, 2022-05-17:
These are hooks to easily superclass the HaClient and implement custom logic if needed (i.e: monitoring). I'd recommend to just define these functions with pass to appease the type checker.

Let me know if you need my help with this!

PR #908 (2022-06-06): Reconciliator: added function definitions
- help for type check
- highlight function arguments types

Closes #885
commit d39a56d8 (2022-06-06): Reconciliator: added function definitions

- help for type check
- highlight function arguments types

Closes #885
closed 2022-06-06 by oroulet