FreeOpcUa/opcua-asyncio #899: Unknown datatype for field Unknown data - Not handling type word correctly

jmtatsch, 2022-05-20:
When I initially connect to a Siemens S7 using the gui it dies due to this:

```
uawidgets.utils - ERROR - Unknown datatype for field: StructureField(Name='awInt', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=3002, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=1, ArrayDimensions=[3], MaxStringLength=0, IsOptional=False) in structure:_some_structure_, please report')
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/home/user/.local/lib/python3.8/site-packages/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/home/user/.local/lib/python3.8/site-packages/uaclient/uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 444, in result
    return self.__get_result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 676, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 324, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 242, in _generate_object
    code = make_structure_code(data_type, name, sdef)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 192, in make_structure_code
    raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")
```

**Describe the bug** <br />
asyncua is unable to handle an array of datatype word within a struct and raises a runtime error

This is the array of words inside a structure it chokes on:

[image]

**Expected behavior**<br />
Should not raise a runtime error but handle the datatype word correctly.

**Version**<br />
Python-Version: 3.8<br />
opcua-asyncio: 0.9.92:

schroeder-, 2022-05-21:
I think we have some issues resolving types in the correct order, i will try to work it out soon. 
In the mean time, you could try load_type_definitions instead.

oroulet, 2022-05-21:
I am rather sure load_type_definition is more broken than load_data_type_definition. But yes somehow we need to order dependencies

schroeder-, 2022-05-22:
@jmtatsch can you try my draft #901  and tell me if I fixed all errors?

jmtatsch, 2022-05-22:
First of all, I really appreciate you looking into that. 

I am using opcua-asyncio via the gui when this happens.
As far as I can tell both your proposed calls should happen upon connect()
https://github.com/FreeOpcUa/opcua-client-gui/blob/cd96cc60e783c8f91b0cecd64573d296c2f94d0c/uaclient/uaclient.py#L126

So you want me to call load_type_definitions first?
Then this happens:

```
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
asyncua.common.structures - ERROR - Failed to execute auto-generated code from UA datatype: 

@dataclass
class FetchResultDataDataType:

    '''
    FetchResultDataDataType structure autogenerated from xml
    '''

    SequenceNumber:ua.Int32 = 0
    EndOfResults:ua.Boolean = True
    ParameterDefs:List[ua.ParameterResultDataType] = field(default_factory=list)
')
Traceback (most recent call last):
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/common/structures.py", line 281, in _generate_python_class
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 12, in FetchResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - module 'asyncua.ua' has no attribute 'ParameterResultDataType'')
Traceback (most recent call last):
  File "/home/jtatsch/.local/lib/python3.8/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/home/jtatsch/.local/lib/python3.8/site-packages/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/home/jtatsch/.local/lib/python3.8/site-packages/uaclient/uaclient.py", line 127, in connect
    self.client.load_type_definitions()
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 444, in result
    return self.__get_result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 668, in load_type_definitions
    return await load_type_definitions(self, nodes)
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/common/structures.py", line 223, in load_type_definitions
    generator.get_python_classes(structs_dict)
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/common/structures.py", line 177, in get_python_classes
    return _generate_python_class(self.model, env=env)
  File "/home/jtatsch/.local/lib/python3.8/site-packages/asyncua/common/structures.py", line 281, in _generate_python_class
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 12, in FetchResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
uaclient.mainwindow - WARNING - showing error: %s')
```
I will look into your draft in a minute.

jmtatsch, 2022-05-22:
With your fixes I am able to browse the server.
Thanks a million!

slokhorst, 2022-08-28:
This fix is released in [v0.9.95](https://github.com/FreeOpcUa/opcua-asyncio/releases/tag/v0.9.95), right?

schroeder-, 2022-08-28:
Should be fixed in v0.9.95 otherwise there is another bug.

closed 2022-08-28 by schroeder-