FreeOpcUa/opcua-asyncio #904: Why does subscribing to child events process through its parent for attributes?

deeTEEcee, 2022-05-26:
I was looking at the logic for how event attributes get added when you subscribe to them and don't quite understand why the child event includes the attributes of it's parents. 

Isn't the other way around (Parent including the attributes of all its children) more intuitive? 
This seems like a bug to me

schroeder-, 2022-05-26:
The current behavior you describe is correct. A Event is always derived from a parent class, at least from BaseEvent. So you always want all Fields from the parent and non subtypes of the Events child nodes.

oroulet, 2022-05-26:
That is inheritance per definition. Many strange stuff in UA but that one seems correct

deeTEEcee, 2022-05-26:
hmm, let me find an example and share it here.

deeTEEcee, 2022-05-27:
Okay, I just want to make sure I understand.

Let's say we have the following event structure (This is how it would appear in UAExpert):
```
Event1
   * attr1
   * attr2
   * attr3
   * Event1.NestedEvent
      * attr4
      * attr5
Event2
...
```

From my view, when I subscribe to `Event1.NestedEvent`, I would only read attr4 and attr5 but currently, it would include the attributes of its parent `Event1` as well. While that seems counter-intuitive to me, are we saying this is expected and how OPC-UA should operate?

deeTEEcee, 2022-05-27:
Thanks for the confirmation. I'll go ahead and close this ticket.

> you can find the spec online and read it if you want

Thanks, I'll give it a try but I don't think understanding it is going to be that simple.

closed 2022-05-27 by deeTEEcee

deeTEEcee, 2022-06-02:
OPCUA specs aside, I think the issue I mentioned becomes much clearer if you looked at this in UAExpert. In UAExpert, you can mark checkboxes to enable events+attributes.

When you subscribe to a child event by clicking on the checkbox for it, the attributes of the parent will not be marked. With the opcua code, those get included by default.