FreeOpcUa/opcua-asyncio #905: Add event support for nested objects (NodeClass.Object)

deeTEEcee, 2022-06-03:
I see that in https://github.com/FreeOpcUa/opcua-asyncio/pull/415 we added support for variables with properties but it looks like we're still missing `NodeClass.Object` in that filter. These can then contain their own properties, variables, or nested objects.

closed 2022-06-05 by oroulet

deeTEEcee, 2022-06-06:
Thanks for making the change. A side note is we're calling `get_browse_name()` quite a bit with the Node class. I wonder if it makes sense just to cache browse name as part of the node. 

For example, I was subscribed to 1 event which contained about 400 nodes and that took 10 minutes to load. In cases like https://github.com/sightmachine/opcua-asyncio/blob/master/asyncua/common/node.py#L414-L425, we already have the browse name available and passing that into Node ahead of time, the time cut in half.

schroeder-, 2022-06-06:
There are a lot room of optimization in this code, but you are correct we have the browse_names already. I can look into using the low level api and speed up the code.

For your use case you could execute the code once and pickle the EventFilter that is created and reuse it for the next time.

I don't know if a event with 400 nodes something that you want to have. Sounds like something that should't be an event.

deeTEEcee, 2022-06-06:
> I don't know if a event with 400 nodes something that you want to have. Sounds like something that should't be an event.

🤔 Not sure. From a data perspective, 400 data points isn't that much in the large scope of a manufacturing plant. Whether that setup actually makes sense, that would be something I'd have to ask the customer about.

deeTEEcee, 2022-06-06:
It's definitely overkill for us right now because I setup application-side filtering after fetching the default `subscribe_events` call and we currently do only need a few pieces. The opcua lib was new to me and the interface is not very straightforward so I ended up doing it this way to start.

schroeder-, 2022-06-07:
```Python
filter = get_filter_from_event_type([NodeOfTheEvent])
# pickle filter
sub.subscribe_events(... evfilter=filter)
```

> It's definitely overkill for us right now because I setup application-side filtering after fetching the default subscribe_events call and we currently do only need a few nodes. 

Then create your own filter (Example is for AlarmConditionEvents):

```Python
content_filter = ContentFilter(
    [
        ContentFilterElement(
            FilterOperator.OfType,
            [LiteralOperand(Variant(NodeId(ObjectIds.AlarmConditionType)))],
        )
    ]
)
filter = EventFilter(
    [
        SimpleAttributeOperand(
            BrowsePath=[QualifiedName("EventType")],
            AttributeId=AttributeIds.Value,
            TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
        ),
        SimpleAttributeOperand(
            BrowsePath=[QualifiedName("Message")],
            AttributeId=AttributeIds.Value,
            TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
        ),
        SimpleAttributeOperand(
            BrowsePath=[QualifiedName("ActiveState"), QualifiedName("Id")],
            AttributeId=AttributeIds.Value,
            TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
        ),
        SimpleAttributeOperand(
            BrowsePath=[QualifiedName("ConditionName")],
            AttributeId=AttributeIds.Value,
            TypeDefinitionId=NodeId(ObjectIds.BaseEventType),
        ),
    ],
    WhereClause=content_filter,
)

```

PR #906 (2022-06-05): EventFilter: handle objects in EventTypes
If a EventType has a object as child, add it's variables and properties to the eventfilter. Also handles variables that contain nested variables or objects.
Fixes #905