FreeOpcUa/opcua-asyncio #925: ServerTimestamp not updated when data are written from outside

jfrechard, 2022-06-10:
I have a very simple OPC-UA server, a "blackbox" client send data to this server and everything is working fine except the timestamp.

When I add value to a variable using set_value, the source timestamp is updated. However when  the client write data value, the Source Timestamp is replaced by 01:00:00:000.

The Server timestamp stay at the value (01:00:00:000) whatever I write value from the python code or from outside.

Is there a way to update the Server Timestamp or source timestamp when values are written from outside the python code ?

schroeder-, 2022-06-10:
We do nothing with Timestamps currently, what ever Timestamps are set in the DataValue are used. So if a client doesn't send Server/SourceTimestamp we set it to (NULL) 01:00:00:0000.

jfrechard, 2022-06-15:
Thank you for answer !  

Maybe I will add a pull request on the subject.

closed 2022-06-15 by jfrechard