FreeOpcUa/opcua-asyncio #926: DataType syntax

om327, 2022-06-10:
Hello, 

I've been working on integrating some asyncio servers with LabVIEW's SystemLink OPC UA plug-in. 

Following a couple of service requests we've noticed that there is a slight difference how DataType is represented. In the older opcua-python library this is represended as, for example 'String' or 'Double'.  The asyncio version seems to display DataType as an ENUM for example i=12 (string). Is there a reason for this change and is it possibel tto enforce a particular format for DataType as it seems to prevent some clients from connecting?

Many thanks!

schroeder-, 2022-06-10:
There isn't any change, DataTypes are always NodeIds, there are no Enums behind it. Maybe you have a minimal example?

om327, 2022-06-12:
I've tried to use the same commands for both implimentations but i'm still getting the same DataType difference when using the asyncio (DataType i=11) rather than DataType=Double in the client application. In UaExpert is doens't seem to make a difference and in both instances DataType is shown as Double.

python-opcua
```
import time
from opcua import ua, Server

if __name__ == "__main__":

    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    myobj = server.nodes.objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()

    server.start()
    
    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            myvar.set_value(count)
    finally:
        server.stop()
``` 

opcua-ascyncio
```
import asyncio
from asyncua import ua, Server

async def main():
    server = Server()
    await server.init()
    server.set_endpoint('opc.tcp://0.0.0.0:4840/freeopcua/server/')

    uri = 'http://examples.freeopcua.github.io'
    idx = await server.register_namespace(uri)

    myobj = await server.nodes.objects.add_object(idx, 'MyObject')
    myvar = await myobj.add_variable(idx, 'MyVariable', 6.7)
    await myvar.set_writable()
    async with server:
        while True:
            await asyncio.sleep(1)
            new_val = await myvar.get_value() + 0.1
            await myvar.write_value(new_val)

if __name__ == '__main__':
    asyncio.run(main())
```

oroulet, 2022-06-13:
what is you see in the UI does not mean anything  i=11 and Double are the same datatype. But it is a bit strange that what is shown in UI is differnet between python-opcua and opcua-asyncio. Something is confusing them. Maybe if you monitor the network with datashark you can find a difference when readong a simple double variable? maybe  we are not encoding the nodeid the same way...

schroeder-, 2022-06-16:
If compared the differences between python-opcua and asyncua and found the following:
[image]
[image]

I will make a fix to address the difference.

oroulet, 2022-06-16:
if that is true, there is something strange. DataTypeDefinition should only be set for ObjectTypes. No  idea where the bug is but this must somewhere very deep

schroeder-, 2022-06-16:
The default is an empty ExtensionObject. We must return a error AttributeIdInvalid if the ExtensionObject is empty. 
This change triggers some errors in some other code that does expect empty ExtensionObjects instead of StatusCode errors...

oroulet, 2022-06-16:
OK looks like your screenshot is for DataType under Types/DataTypes. Great you are looking at it. Did you check the spec for the correct behaviour?

schroeder-, 2022-06-16:
@om327  can you test it with fix #937 ?
If it works we can merge it, if it doesn't help I will do some more research about the spec.

oroulet, 2022-06-16:
@schroeder- Aren't the clients more happy with None instead of error? I do not remember if we can set an ExtensionObject to None, but we probably can. Maybe that is what python-opcua does?

om327, 2022-06-16:
How do i use the command ```gh pr checkout 937``` to checkout the new code?

Does this need to part of a ```git fetch origin``` statment? 

Sorry my knowledge of git is little vague beyond cloneing, pulling and pushing!

oroulet, 2022-06-16:
> Ok DataTypeDefinition is optional as of https://reference.opcfoundation.org/v104/Core/docs/Part3/5.8.3/.
> So return BadAttributeIdInvalid is ok if it doesn't exists or is empty ExtensionObject.

BadAttributeInvalid is  fine if attribute is invalid. but I am wondering if creating empty ExtensionObject is a good idea at all in any places... maybe the default should be None instead of empty EntensionObject. Extension objects seems to be nullable: https://reference.opcfoundation.org/v105/Core/docs/Part6/5.1.2/

In that particular case the attribute should probably do not exist for these typoes apart for structs and enums

om327, 2022-06-16:
Ah found it - if in doubt just read the manual!!! 

Just waiting for NI to update our license and then will be able to test out! If you want to have two versions of the attribute i can test both to see if they are both compatable?

emiliodiritaRA, 2022-10-17:
> In that particular case the attribute should probably do not exist for these typoes apart for structs and enums

I agree on this.

I recently had a similar problem, and I think it may be related to this: https://github.com/FreeOpcUa/opcua-asyncio/issues/1082

In fact, the existence of the attribute (even if it returns BadAttributeIdInvalid), I think is incorrect. The attribute is mandatory for structure or Enumeration datatypes, so I think it is optional "for other DataTypes".
Finding it on types, as it happens now, I think is wrong.

emiliodiritaRA, 2023-01-19:
Any update on this?

schroeder-, 2023-01-19:
I quickly checked against a server with Softing SDK and I found we are missing a reference.
[image]. This has to be fixed in the schema generator.
Also in OPC UA there is nothing like a list of available attributes. The client decides which attributes to read, based on the nodetype and references. The server can only return BadAttributeInvalid or a concret value.

adriantom, 2023-02-13:
How's the work going? Any progress?