FreeOpcUa/opcua-asyncio #941: asyncua.common.xmlimporter: failure adding node NodeData

jmtatsch, 2022-06-20:
Hello,

I have an OPC.xml dumped from a Siemens s7 I would like to load, so I can simulate it and work on it.
When I try to load it looks okay except for that one warning:

```
...
INFO:asyncua.common.xmlimporter:Importing xml node (QualifiedName(NamespaceIndex=2, Name='REMOTE'), NodeId(Identifier=3066, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) as (QualifiedName(NamespaceIndex=2, Name='REMOTE') NodeId(Identifier=3066, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>))
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=3066, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>))
WARNING:asyncua:"The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
INFO:asyncua:Starting server!
WARNING:asyncua.server.server:Endpoints other than open requested but private key and certificate are not set.
INFO:asyncua.server.internal_server:starting internal server
INFO:asyncio:<Server sockets=(<asyncio.TransportSocket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6, laddr=('127.0.0.1', 4840)>,)> is serving
INFO:asyncua.server.binary_server_asyncio:Listening on 127.0.0.1:4840
DEBUG:asyncua.server.server:OPC UA Server(opc.tcp://127.0.0.1:4840) server started
```

But when I inspect the contents on the server with the opcua-client I cannot find any of the defined nodes/variables etc.

**To Reproduce**<br />
My XML is imported like this:
```
async def main():
    server = Server()
    await server.init()
    await server.import_xml("OPC.xml")
    server.set_endpoint('opc.tcp://127.0.0.1:4840')
    async with server:
        while True:
            await asyncio.sleep(1)
```

In the xml node 3065 got parsed just fine but 3066 doesn't have a references node.
Maybe that causes the warning and cancels the parsing altogether?
```
  <UADataType NodeId="ns=1;i=3065" BrowseName="1:ANY">
    <DisplayName>ANY</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype">ns=1;i=3066</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=3</Reference>
    </References>
  </UADataType>
  <UADataType NodeId="ns=1;i=3066" BrowseName="1:REMOTE">
    <DisplayName>REMOTE</DisplayName>
  </UADataType>
```

**Expected behavior**<br />
I expect to see the same nodes/variables as defined in the s7.

**Version**<br />
Python-Version: 3.10
opcua-asyncio Version: master
OPC.xml generated by Generator Product="TIA Portal" Edition="STEP 7 - Server Interface Generation" Version="V1.0.0"

schroeder-, 2022-06-20:
You can't add a node without a parent to the address space. So the warning is correct.

schroeder-, 2022-06-21:
If you remove the following lines, does it work?
```
<UADataType NodeId="ns=1;i=3066" BrowseName="1:REMOTE">
    <DisplayName>REMOTE</DisplayName>
  </UADataType>
```
Because if this node is insert, the xml parsing is stopped and not further nodes are added.

You can ignore this warning in opcua-client, its ok. load_data_type_definitions must be called for pre 1.04 clients.
```
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
```

wvalcke, 2022-06-21:
The Siemens stuff is broken, see also this link https://support.industry.siemens.com/forum/WW/en/posts/siome-xml-export/221307
I got the same issues using SIOME tool from Siemens, generating an XML definition and trying to import in asyncua. You need to patch the xml  then it will work.

jmtatsch, 2022-06-21:
So removing these nodes manually gets me a bit further every time. But I gave up after 20 time restarting, editing, restarting editing.... Any hints how to patch an entire file with MANY of these ?

jmtatsch, 2022-07-07:
It loads with just these warnings now:
```
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
```
But doesnt show the expected objects in the viewer.

elmovw-evac, 2022-07-13:
The parent node id does not to refer to a valid node.

I think I've resolved this problem in my fork. The problem was that in `structures104`, data type dependencies are not sorted correctly. I'll make a PR for it now.

jmtatsch, 2022-07-16:
I can confirm that this vastly improves parsing the xml file.

Only on like the last line I still get that issue:
```
INFO:asyncua.common.xmlimporter:Importing xml node (QualifiedName(NamespaceIndex=2, Name='OB_CYCLIC'), NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) as (QualifiedName(NamespaceIndex=2, Name='OB_CYCLIC') NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>))
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
WARNING:asyncua:"The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
```
Nevertheless i still cannot browse the nodes i would expect to see.

Also a bunch of 
`INFO:asyncua.common.xmlparser:Not implemented tag: <Element '{http://opcfoundation.org/UA/2011/03/UANodeSet.xsd}Extensions' at 0x7fb522dcac00>` became visible now. But i guess that's a different issue.

closed 2022-07-17 by oroulet

jmtatsch, 2022-07-18:
Unfortunately  the current master still causes a single one of the original errors:

```
INFO:asyncua.common.xmlimporter:Importing xml node (QualifiedName(NamespaceIndex=2, Name='OB_CYCLIC'), NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) as (QualifiedName(NamespaceIndex=2, Name='OB_CYCLIC') NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>))
WARNING:asyncua.common.xmlimporter:failure adding node NodeData(nodeid:NodeId(Identifier=3051, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
WARNING:asyncua:"The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
```

Any idea why that is the case @elmovw-evac ?

PR #965 (2022-07-17): structures104: sort type dependencies correctly
This should fix #941, and other cases of importing XML nodesets with custom data structures.

The original problem was that each node would only be compared with its direct dependencies, instead of considering the whole type dependency chain.