FreeOpcUa/opcua-asyncio #952: Unable to find all nodes by ua_utils.get_node_children()

omidfathi, 2022-06-27:
Hi
I want to connect to Prosys opcua server simulation, and i need to find all nodes on server.so i use ua_utils.get_node_children(client.nodes.objects) but I get this error:
_'NoneType' object has no attribute 'send_request'_

//////////////////////////////
`import logging
import asyncio

from asyncua import Client , Node ,ua
from asyncua.common import ua_utils
async def main():
    client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
    client.session_timeout = 2000
    root = client.nodes.root
    while True:
            allVariables = []
            node_List = await ua_utils.get_node_children(client.nodes.objects)
            for i in node_List:
                nodeClass = await i.read_node_class()
                if nodeClass == ua.NodeClass.Variable:
                        allVariables.append(n)
            print(allVariables)
            await asyncio.sleep(1)

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    asyncio.run(main())
`
//////////////////////////////

[image]

schroeder-, 2022-06-28:
I tested get_node_children against our server it is working correct. 
As I don't have access to a Prosys OPC UA Simulation Server, can you provide a extended stack trace or debug log?

schroeder-, 2022-06-28:
Also i noticed your screenshot doesn't match you posted code. Is `objects == client.nodes.objects` ?

omidfathi, 2022-06-28:
> Also i noticed your screenshot doesn't match you posted code. Is `objects == client.nodes.objects` ?

yes, `objects = client.nodes.objects`

schroeder-, 2022-06-28:
Ok the problem is you did not connect to the server.

```Python
async def main():
    client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
    client.session_timeout = 2000
    root = client.nodes.root
    async with client:
      allVariables = []
      node_List = await ua_utils.get_node_children(client.nodes.objects)
      for i in node_List:
          nodeClass = await i.read_node_class()
          if nodeClass == ua.NodeClass.Variable:
                  allVariables.append(n)
      print(allVariables)
      await asyncio.sleep(1)
```

omidfathi, 2022-06-28:
> I tested get_node_children against our server it is working correct. As I don't have access to a Prosys OPC UA Simulation Server, can you provide a extended stack trace or debug log?

This is debug log:
[image]

omidfathi, 2022-06-28:
> I tested get_node_children against our server it is working correct. As I don't have access to a Prosys OPC UA Simulation Server, can you provide a extended stack trace or debug log?

This is debug log:
[image]

> Ok the problem is you did not connect to the server.
>
> ```python
> async def main():
>     client = Client("opc.tcp://localhost:53530/OPCUA/SimulationServer/")
>     client.session_timeout = 2000
>     root [...]

What a clear mistake !!!
Sorry to take your time ...
It's work now.

omidfathi, 2022-06-29:
Hello again

There is a new problem, to collecting the values an error happened "asyncua.common.utils.NotEnoughData"
[image]

schroeder-, 2022-06-29:
The decoding of a value fails, maybe it is needed to load custom datatypes:
`await client.load_data_type_definitions()`

omidfathi, 2022-06-29:
> The decoding of a value fails, maybe it is needed to load custom datatypes: `await client.load_data_type_definitions()`

Still same error ...

AiyionPrime, 2025-03-25:
Well. I encountered the very same error @omidfathi.
`NotEnoughData: Not enough data left in buffer, request for 1426063360, we have 78`.

asyncua==1.1.5 @schroeder- is there any suggestion on how to debug it?

AiyionPrime, 2025-03-25:
I run against open62541 1.4.6 as shipped in [open62541pp from conan2](https://conan.io/center/recipes/open62541pp)

oroulet, 2025-03-25:
that is typically a parsing error. You need to identify which node leads to that issue

AiyionPrime, 2025-03-25:
@AndreasHeine The node is indeed an ExtensionObject.

AiyionPrime, 2025-03-25:
@oroulet Server.ServerDiagnostics.SessionDiagnosticsSummary.[client_name].SessionSecurityDiagnostics and the value of that.

oroulet, 2025-03-25:
also are you using the new load_data_type_definiton or the old method? the old one has many many limitations and the new one probably is still missing some data types but at least code is much easier to read and improve...
getting the data type definition and comparing that to what is supported in code is probably the first step

AiyionPrime, 2025-03-25:
```python
url = "opc.tcp://localhost:48401/"

from asyncua.sync import Client
from asyncua import ua
import pdb

# connect to the opcua server
with Client(url) as client:
    client.load_data_type_definitions() # this does not appear to make a difference?
    # Get the node
    nodeid = ua.NodeId(55204, 1)
    node = client.get_node(nodeid)

    pdb.set_trace()

    # Get the node value
    #value = node.get_value()
```

oroulet, 2025-03-25:
call node.read_type_definition() (or very similar named method on the node object) and report here

AiyionPrime, 2025-03-25:
```console
(Pdb) dir(node)
['_SyncNode__get_nodeid', '_SyncNode__set_nodeid', '__annotations__', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'add_data_type', 'add_folder', 'add_method', 'add_object', 'add_object_type', 'add_property', 'add_reference', 'add_reference_type', 'add_variable', 'add_variable_type', 'aio_obj', 'call_method', 'delete', 'delete_reference', 'get_access_level', 'get_child', 'get_children', 'get_children_by_path', 'get_children_descriptions', 'get_data_type_as_variant_type', 'get_data_value', 'get_description_refs', 'get_encoding_refs', 'get_methods', 'get_parent', 'get_path', 'get_properties', 'get_referenced_nodes', 'get_references', 'get_user_access_level', 'get_value', 'get_variables', 'history_read', 'history_read_events', 'nodeid', 'read_array_dimensions', 'read_attribute', 'read_attributes', 'read_browse_name', 'read_data_type', 'read_data_type_as_variant_type', 'read_data_type_definition', 'read_data_value', 'read_description', 'read_display_name', 'read_event_history', 'read_event_notifier', 'read_node_class', 'read_params', 'read_raw_history', 'read_type_definition', 'read_value', 'read_value_rank', 'register', 'set_attr_bit', 'set_event_notifier', 'set_modelling_rule', 'set_value', 'set_writable', 'tloop', 'unregister', 'unset_attr_bit', 'write_array_dimensions', 'write_attribute', 'write_data_type_definition', 'write_params', 'write_value', 'write_value_rank']
```

`read_type_definition` exists and is:

`<bound method syncmethod.<locals>.wrapper of SyncNode(NodeId(Identifier=55204, NamespaceIndex=1, NodeIdType=<NodeIdType.FourByte: 1>))>`

Invoking it yields:
`NodeId(Identifier=2244, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>)`

AndreasHeine, 2025-03-25:
its "SessionSecurityDiagnosticsType"

AiyionPrime, 2025-03-25:
https://reference.opcfoundation.org/nodesets/162/89023

AiyionPrime, 2025-03-25:
I added this in the for-loop, which resolves the problem at hand, but not the underlying issue.

```python
if b_name.Name == "SessionSecurityDiagnostics":
                    # This is a hotfix for a broken value reading.
                    # See https://github.com/FreeOpcUa/opcua-asyncio/issues/952 for details.
                    continue
```

AiyionPrime, 2025-03-25:
A brief search in open62541 yields this, which might be related

https://github.com/open62541/open62541/issues/6291

AiyionPrime, 2025-03-25:
That one was not related, I'd say. It has been merged in February "24 and the version I'm using is from October "24.

Looking at the path I posted above: Is the server reporting some kind of reflection of the client?

So is this possibly an asyncua problem, despite coming *from the server* open62541?

oroulet, 2025-03-26:
type definition is here: https://reference.opcfoundation.org/nodesets/156/60183
there is really nothing special here. So someone should sit down and debug what the parser does here in structure104.py. Really not sure I will get time anytime soon

AiyionPrime, 2025-03-26:
I think we can safely remove the "fixed?" label :)