FreeOpcUa/opcua-asyncio #960: SecurityPolicyBasic256: confusing argument name

brubbel, 2022-07-10:
https://github.com/FreeOpcUa/opcua-asyncio/blob/0f4f68cdaee8a5a9ba193272876a1c7ac66937eb/asyncua/crypto/security_policies.py#L555

`client_pk` is in fact the private key for the host certificate and as such `privateKey` should be a better name.

schroeder-, 2022-07-17:
The code is used in client and server. So it should just named privatekey or pk.

brubbel, 2022-08-03:
Given that in the context of the client, the `host_cert` and `client_pk` are still paired, I do not agree.
Maybe the use of `local_cert` and `local_pk` is better, along with `peer_cert`.

schroeder-, 2022-08-03:
I didn't touch all that cryto stuff, if you can provide a pr with `local_cert` and `local_pk` this would be nice. Also this is the same for other SecurityPolicys.

starturtle, 2022-08-17:
I find "local" and "host" equally confusing.
Basically, we're talking about the client application, client user, and server application certificates when encrypting messages and doing certificate-based authentication. 
Imagine a client that's located on the same system as the server. Then "local" and "host" are the same system. 
What is "peer" in that context? The server?
So far, using "server", "client" and "user" has proven useful. They're near impossible to confuse.
Anyhow, having the same name for key and certificate is definitely _much_ more readable, no matter the actual name. 👍