FreeOpcUa/opcua-asyncio #962: NodeID string is truncated

WOLF33, 2022-07-12:
My server uses the following identifier "ns=3;s=EquipSetParam(E=93;SP=204)" to identify object parameters. when I get a node, the identifier is truncated to something like this "ns=3;s=EquipSetParam(E=93", so I can't get the node and data. I looked at the source code of the package, there is a semicolon separation. What should I do to start get data by my IDs?

schroeder-, 2022-07-12:
As workaround you can try `ua.NodeId("NodeIdstrWith;", idx)`

WOLF33, 2022-07-12:
I took a different path, took the source code of the class generation as a basis and introduced a limiter equal to 1 into the split method
[image]