FreeOpcUa/opcua-asyncio #967: xmlparser cannot parse TIA Portal xml due to Not implemented tag: <Element '{http://opcfoundation.org/UA/2011/03/UANodeSet.xsd}Extensions'

jmtatsch, 2022-07-16:
When importing a TIA Portal generated xml file the following error appears for like every node:

```
INFO:asyncua.common.xmlparser:Not implemented tag: <Element '{http://opcfoundation.org/UA/2011/03/UANodeSet.xsd}Extensions' at 0x7fb522dcac00>
```
Maybe as a result of this these nodes are not visible for browsing with the gui.

opcua-asyncio Version : master branch

schroeder-, 2022-07-17:
To support ExtensionObjects or custom python structs in xml, we need to implement the whole data encoding for xml (https://reference.opcfoundation.org/v105/Core/docs/Part6/5.3.1/). This not a trivial task, see https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/ua/ua_binary.py which implements the binary encoding.

schroeder-, 2023-03-29:
Maybe with #1259 and a lot of other changes it works?

closed 2023-03-31 by schroeder-