FreeOpcUa/opcua-asyncio #972: ThreadLoop starts and runs only once

HeytalePazguato, 2022-07-21:
I am trying to use asyncua.sync but the ThreadLoop runs only once, why is that? (I am using version 0.9.94)

I have made a simple test with an OPC UA server that toggles a boolean variable (outBOOL) every second, I found the examples for having a sync Client and made a simple code to be able to read the `outBOOL` variable

My expectation is that the value will be printed every second and therefore show the True, False, True, False, ... sequence but somehow after the first time it stops and executes the line `print('tloop stopped')`, how can I keep the loop running and therefore print the value of the node?

```
import time
from IPython import embed
from IPython.display import clear_output
from asyncua.sync import Client, ThreadLoop

# Create cycle
def sleep(duration, get_now=time.perf_counter):
    now = get_now()
    end = now + duration
    while now < end:
        now = get_now()
        
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

# Environment variables
ip = '127.0.0.1'
port = '4840'

with ThreadLoop() as tloop:
    with Client(f'opc.tcp://{ip}:{port}', tloop = tloop) as client:
        clear_output(wait=True)
        st = time.time()
        print(client.tloop, client.close_tloop)
        client.load_data_type_definitions()
        
        node = client.get_node('ns=6;s=::testOPCUA:outBOOL')
        print(node.read_value())
        
        sleep(0.5-(time.time() - st))

print('tloop stopped')
```

If I create a subscription and `subscribe_events` also doesn't work, unless I use the `embed()` function, why is that? Is there no way to keep the thread running?

This works:
```
import time
from IPython import embed
from IPython.display import clear_output
from asyncua.sync import Client, ThreadLoop

# Create cycle
def sleep(duration, get_now=time.perf_counter):
    now = get_now()
    end = now + duration
    while now < end:
        now = get_now()
        
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

# Environment variables
ip = '127.0.0.1'
port = '4840'

#inBOOL = PVNode(client,'ns=6;s=::testOPCUA:outBOOL', 'BOOL')

with ThreadLoop() as tloop:
    with Client(f'opc.tcp://{ip}:{port}', tloop = tloop) as client:
        clear_output(wait=True)
        st = time.time()
        print(client.tloop, client.close_tloop)
        client.load_data_type_definitions()
        
        node = client.get_node('ns=6;s=::testOPCUA:outBOOL')
        
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(node)
        
        sleep(0.5-(time.time() - st))
        sub.subscribe_events()
        
        embed() #How to make it work without the embed function?
print('tloop stopped')
```

schroeder-, 2022-07-21:
Python has the gil, so your sleep function prevents any code from running, because you burn cpu time. Use time.sleep instead if you don't want to block any other code.

Also ThreadLoop is not a repeating loop change your code to this:

```
with ThreadLoop() as tloop:
    with Client(f'opc.tcp://{ip}:{port}', tloop = tloop) as client:
        clear_output(wait=True)
        st = time.time()
        print(client.tloop, client.close_tloop)
        client.load_data_type_definitions()
        
        node = client.get_node('ns=6;s=::testOPCUA:outBOOL')
        
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(node)
        
        time.sleep(0.5-(time.time() - st))
        sub.subscribe_events()
        while 1:
             time.sleep(0.5)
             #Do something
```

oroulet, 2022-07-21:
> Python has the gil, so your sleep function prevents any code from running, 

What?

The issue over is imply the lifetime behavior of `with` when you get out of ẁith` block then the ThreadLoop and Client die/disconnect, so of course you do not get any subscrption. But the code snippset from @schroeder- is correct. ;-) If that does nto work then do not use `with`and diconnect/connect manually

schroeder-, 2022-07-21:
His sleep code is just burning cpu time so no other thread can be executed.

oroulet, 2022-07-21:
oh shit he is writing his own sleep method! do not do that

HeytalePazguato, 2022-07-25:
Thanks for the prompt reply, it was not the `sleep` method, perhaps the missing `while 1` that I changed to `True`

```
import time
from IPython import embed
from IPython.display import clear_output
from asyncua.sync import Client, ThreadLoop

# Create cycle
def sleep(duration, get_now=time.perf_counter):
    now = get_now()
    end = now + duration
    while now < end:
        now = get_now()
        
class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

# Environment variables
ip = '127.0.0.1'
#ip = '10.81.10.114'
port = '4840'

#inBOOL = PVNode(client,'ns=6;s=::testOPCUA:outBOOL', 'BOOL')

with ThreadLoop() as tloop:
    with Client(f'opc.tcp://{ip}:{port}', tloop = tloop) as client:
        
        client.load_data_type_definitions()
        
        node = client.get_node('ns=6;s=::testOPCUA:outBOOL')
        
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(node)
        
        sub.subscribe_events()
        while True:
            st = time.time()
            clear_output(wait=True)
            sleep(1-(time.time() - st))
             #Do something
                
print('tloop stopped')
```

This works properly, it was not the sleep method.

closed 2022-07-25 by HeytalePazguato

oroulet, 2022-07-25:
The sleep is still missing of it does not killed your program now it will later. Check your cpu use.
And never reuse well known method names like sleep. It will confuse every reader