FreeOpcUa/opcua-asyncio #974: Raise exception of _renew_channel_task correctly

elboerto, 2022-07-27:
**Describe the bug** <br />
`async def _renew_channel_loop(self)` and the related task created in `async def create_session(self):` does not raise Exceptions correctly. If any exceptions occur within the _renew_channel_loop task, the client stops working but does not crash.

**To Reproduce**<br />
Throw an exception in the "while True" loop of `_renew_channel_loop(self)`

**Expected behavior**<br />
The client should crash which would enable the user to handle it. The exception of the created task `_renew_channel_task` should be raised by e.g. calling `_renew_channel_task.result()`

**Screenshots**<br />
[image]

**Version**<br />
Python-Version: 3.9.13
opcua-asyncio Version (e.g. master branch, 0.9): master branch, 0.9.94

oroulet, 2022-07-27:
PR welcome if you have an idea for an API

schroeder-, 2022-09-20:
Fixed in #986

closed 2022-09-20 by schroeder-