FreeOpcUa/opcua-asyncio #989: Should we switch returnBounds to default to False instead?

deeTEEcee, 2022-08-05:
Is there a good reason [DefaultBounds](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/common/node.py#L523) is set to True by default?

It seems more intuitive to set it to False. I don't see why people would stream a time range and expect data outside that range.

deeTEEcee, 2022-08-05:
In my case, we run this to stream data from a factory and because we're missing monitoring systems for bad data, we missed this. We ended up ingesting a lot of duplicate data as a result of OPC-UA nodes that had no data and as a result would only return returnBounds data infinitely.

oroulet, 2022-08-06:
What part of code are you writing about?

deeTEEcee, 2022-08-08:
> Is there a good reason [DefaultBounds](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/common/node.py#L523) is set to True by default?

Code link in there. Talking about `read_raw_history`

schroeder-, 2022-08-15:
Don't think we should change that. The problem you describe sounds like a no conforming server. See the specs: https://reference.opcfoundation.org/v104/Core/docs/Part11/4.4/

deeTEEcee, 2022-08-15:
I'm saying in my opinion, I don't think the average client user cares about receiving data outside time bounds when running this method. It might surprise you but people don't always read the documentation.

It has nothing to do with a server not following expected behavior.