FreeOpcUa/opcua-asyncio #994: Reading an extension object with nested structures throws NotEnoughData Error

mohitkumaragarwal, 2022-08-10:
Hi,
Can you please suggest your inputs for the below query?

I am trying to use the client API and reading an extension object with nested structures in it. I get Not Enough Data error as shown below. Which configuration parameter can be updated to read a larger blob for the same.

raise NotEnoughData(f"Not enough data left in buffer, request for {size}, we have {self._size}")
asyncua.common.utils.NotEnoughData: Not enough data left in buffer, request for **591983720**, we have **120877**

_Originally posted by @mohitkumaragarwal in https://github.com/FreeOpcUa/opcua-asyncio/discussions/839#discussioncomment-3347668_

schroeder-, 2022-08-10:
Which version are you using? Can you try with the current master?
Also did you check if there are any warnings/errors in the logging outputs?

mohitkumaragarwal, 2022-08-10:
@schroeder- 
Thank you for the quick reply. I am using latest version from master branch given in **https://github.com/FreeOpcUa/opcua-asyncio**

Also, I have attached the following files:

- **python_client_output.txt** - The output of the files 
- **client_to_demo-this.com.py.zip** -- The client code modified to read the structure data.
- **IJT_OPC_UA_Server_Test_Binaries.zip** -- This is the binary of the OPC UA Server which implements OPC 40451-1 CS with simulated data. Adding this in case if we need to troubleshoot the issue.

[python_client_output.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/9300985/python_client_output.txt)
[client_to_demo-this.com.py.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/9301000/client_to_demo-this.com.py.zip)
[IJT_OPC_UA_Server_Test_Binaries.zip](https://github.com/FreeOpcUa/opcua-asyncio/files/9301028/IJT_OPC_UA_Server_Test_Binaries.zip)

mohitkumaragarwal, 2022-08-15:
@schroeder-  Any suggestions on the above issue whenever you are free please?

schroeder-, 2022-08-15:
Thx @mohitkumaragarwal for providing all i need to find the bug. #998 allowed me to read the server data.

mohitkumaragarwal, 2022-08-15:
Thanks a lot @schroeder- for your quick fix. I will pull the latest changes and validate it. Its really great to get the support on this.

PR #998 (2022-08-16): use correct datatype for encoding
EncodingMask was the wrong datatype see: https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.7/
Fixes #994
closed 2022-08-16 by oroulet