FreeOpcUa/opcua-asyncio #999: Failing while Writing value with sourceTimestamp to OPC-UA Server

xsahban, 2022-08-16:
Trying to write the `value` of a tag in OPC-UA Server, with a `sourceTimestamp` with the following method:
```py
def write_opc_tag(url, tag_id, value):
    with Client(url=url) as client:
        node = client.get_node(tag_id)

        dt = ua.DateTime.utcnow().replace(tzinfo=ua.timezone.utc).astimezone(pytz.timezone("US/Eastern"))
        print(dt)

        dv = ua.DataValue(ua.Variant(value, VariantType=ua.VariantType.String),
                          SourceTimestamp=dt)

        node.write_value(dv)

        print(node.read_value())
```

It's failing with the following error:
`asyncua.ua.uaerrors._auto.BadWriteNotSupported: "The server does not support writing the combination of value, status and timestamps provided."(BadWriteNotSupported)`

I want to understand 2 points
1. Can we publish the value with `SourceTimestamp`, if yes and the above code snippet is correct, could it be the configuration issue on the server side?
2. If we publish the value without `SourceTimestamp`, what is the default `SourceTimestamp` getting set, the time of publishing (set by publisher client), or the time when the server receives the value (set by the server itself)?

schroeder-, 2022-08-16:
1. Yes server can deny setting the SourceTimestamp / ServerTimestamp, it is up to the server.
2. Depends on the server, most server will set it to the time when the server receives the value (set by the server itself).

Jusufs, 2023-01-15:
Trying PLCnext AXC F 2152 controller. Issue with time on PLC OPC UA server exists to.
Please help me to write correct code for var,write_value  or var.set_value
 
async def main():
    async with Client("opc.tcp://192.168.100.251:4840/") as client:
        client.set_user('admin')
        client.set_password('cdf0af89')

        root = client.nodes.root
        print("Root node is: ", root)
        objects = client.nodes.objects
        print("Objects node is: ", objects)

# Read PLCnext OPC variable
        var = client.get_node("ns=6;s=Arp.Plc.Eclr/node_number")
        var1 = await var.read_data_value() # get value of node as a DataValue object
        var2 = await var.read_value()      # get value of node as a python builtin
        print("Var1: ",var1)
        print("Var2: ",var2)
??????????????????????????????????????????????
        dv = var.DataValue(5, var.VariantType.UInt16)
        dv.ServerTimestamp = None
        dv.SourceTimestamp = None
??????????????????????????????????????????????????
        await var.set_value(var1)

Answer:
INFO:asyncua.client.ua_client.UaClient:activate_session
Root node is:  i=84
Objects node is:  i=85

Var1:  DataValue(Value=Variant(Value=21, VariantType=<VariantType.Int16: 4>, Dimensions=None, is_array=False), StatusCode_=StatusCode(value=0), SourceTimestamp=datetime.datetime(2023, 1, 15, 18, 33, 54, 485625), ServerTimestamp=None, SourcePicoseconds=None, ServerPicoseconds=None)

Var2:  21
INFO:asyncua.client.client:disconnect
INFO:asyncua.client.ua_client.UaClient:close_session
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.ua_client.UASocketProtocol:Socket has closed connection
Traceback (most recent call last):
  File "/etc/openhab2/scripts/OPC/client.py", line 76, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/etc/openhab2/scripts/OPC/client.py", line 42, in main
    dv = var.DataValue(5, var.VariantType.UInt16)
AttributeError: 'Node' object has no attribute 'DataValue'

schroeder-, 2023-01-15:
Try this:
`dv = ua.DataValue(5, ua.VariantType.UInt16)`

Jusufs, 2023-01-15:
I'm not profi programmer, so how should I substitute dv and ua ?

AndreasHeine, 2023-01-15:
from asyncua import ua

Jusufs, 2023-01-31:
Thank you very much Andreas. I don't know that ua should be imported.

closed 2023-03-29 by schroeder-