FreeOpcUa/opcua-asyncio #1019: OPCUA version not supported "Server doesn't satisfy required XML-Models. Import them first!"

vhavard, 2022-09-06:
**Describe the bug** <br />
The asyncua does not support UA model above 1.04.7.

**To Reproduce**<br />
Create a UA model from UAmodeler, with base version 1.04.7 of UA model http://opcfoundation.org/UA/
Export it to a Nodeset.xml (see attached file)

Run the server 
```python
async def main():
    logging.basicConfig(level=logging.INFO)
    server = Server()
    await server.init()
    # import some nodes from xml
    # http://opcfoundation.org/UA/ version 1.04.7 OK
    nodes1 = await server.import_xml("server-uamodeler-objecttypes-datatype-1_04_7-v4.xml") # OK Objectype with subobject TransformType -> 3DvectorType et QuaternionType
    # http://opcfoundation.org/UA/ version 1.04.11 KO
    # nodes1 = await server.import_xml("server-uamodeler-objecttypes-datatype-1_04_7-v4 - 1_04_11.xml") # KO à cause du numéro de version Objectype with subobject TransformType -> 3DvectorType et QuaternionType
    
    # starting!
    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())
```

it works.

Do the exact same thing with version     http://opcfoundation.org/UA/ version 1.04.11 
It does not work
Error is:  
Exception has occurred: ValueError
Server doesn't satisfy required XML-Models. Import them first!

**Expected behavior**<br />
Being able to use all versions of UA model
Or 
being able to init server with the Opc.Ua.NodeSet2.xml needed for our model.

**Version**<br />
Python-Version:3.10.4<br />
opcua-asyncio Version (e.g. master branch, 0.9): master branch
Files to test (please modify extension to xml):
[server-uamodeler-objecttypes-datatype-1_04_7-v4 - 1_04_11.xml.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/9495022/server-uamodeler-objecttypes-datatype-1_04_7-v4.-.1_04_11.xml.txt)
[server-uamodeler-objecttypes-datatype-1_04_7-v4.xml.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/9495023/server-uamodeler-objecttypes-datatype-1_04_7-v4.xml.txt)

schroeder-, 2022-09-06:
We are currently using http://opcfoundation.org/UA/ version 1.04.10 we haven't updated the library yet. 
In the near future we will update to V1.05.XX.
You can update the schema your self see: https://github.com/FreeOpcUa/opcua-asyncio/tree/master/schemas
Also most of the time the version listed in xml files isn't need. You can just edit the xml in question to 1.04.10 and import it.

vhavard, 2022-09-06:
Thank you for you quick answer.
Is it possible to pass to the UA 1.05.+ since this is the last version of UA version ?

schroeder-, 2022-09-06:
I tried it some time ago and there were some code changes needed to get UA 1.05 working.

vhavard, 2022-09-06:
I tried updating the schemas with https://github.com/FreeOpcUa/opcua-asyncio/tree/master/schemas
The readme.md mention that:
* Path to the .pickle file: opcua-asyncio/asyncua/binary_address_space.pickle
* Path to the .xml Nodesets: opcua-asyncio/schemas/UA-Nodeset-master/Schema/
* Path to the standard_address_space.py files: opcua-asyncio/asyncua/server/standard_address_space/

I was thinking that the files would be updated into my folder. But this is not the case.
Have I done something wrong?

Therefore I tried importing a specific nodeset version of UA with the import_xml command but it does not work
What I don't understand is how to use a specific UA Nodeset.xml in asyncua.
Indeed, if I do that to import a specific version of UA:
```python
nodesua = await server.import_xml("UA-Nodeset-master/Schema/Opc.Ua.NodeSet2.xml")
```
However, I receive this error 
````shell
Exception has occurred: BadNodeIdExists
"The requested node id is already used by another node."(BadNodeIdExists)
  File "D:\CESI\DEV\OPCUA_modeler\test\server-uamodeler-variable.py", line 35, in main
    nodesua = await server.import_xml("D:/CESI/DEV/opcua-asyncio - test update UANodeset/schemas/UA-Nodeset-master/Schema/Opc.Ua.NodeSet2.xml")
  File "D:\CESI\DEV\OPCUA_modeler\test\server-uamodeler-variable.py", line 54, in <module>
    asyncio.run(main())
````

Can you explain me how to update to the right version ?
Thank you

vhavard, 2022-09-07:
> The descrpiton is maybe a bit out of date. You run **update_ua_nodeset.py** that pulls the latest nodeset(Opc.Ua.NodeSet2.xml etc) from github and executes all scripts needed to update the library. This updates the addresspace, objectids, datatypes and many more. For me it did't work on windows, try to use wsl to update the model.
> 
> `nodesua = await server.import_xml("UA-Nodeset-master/Schema/Opc.Ua.NodeSet2.xml")` This is not supported, currently you can only use the Opc.Ua.Nodeset2.xml which is target via schema(update_ua_nodeset.py).

Thank you again for your answer. Indeed, it does not work on Windows for me.
I will try on WSL.
However when I tried on Windows, I notice that the it pulls the latest nodeset(Opc.Ua.NodeSet2.xml etc), which is 1.05.1, which is not supported yet by asyncua as you told me.
Would it be possible to update the scripts to support an update of a specific branch and tag of modest?

schroeder-, 2022-09-22:
Update to 1.05.02 see #1047
Will make a pr to get the schema generation working on windows next.

schroeder-, 2022-09-26:
Schema generation works know under windows in #1049 . Also you can chose now which branch and/or tag to use, via commandline.

closed 2023-03-28 by schroeder-