FreeOpcUa/opcua-asyncio #1023: CVE-2022-25304

schroeder-, 2022-09-07:
For reference this is CVE-2022-25304 and GHSA-mfpj-3qhm-976m

GitHub already raised an alert in our sample server implementation.

_Originally posted by @GoetzGoerisch in https://github.com/FreeOpcUa/opcua-asyncio/issues/1013#issuecomment-1234160597_

kohtala, 2022-09-19:
I think this was fixed in #1039.

schroeder-, 2022-09-19:
No this is in correct. In this dos is because there are no limits on chunk size.
I am working on a fix.

kohtala, 2022-09-19:
So the CVE mentioned on #1013 is incorrect? If you know of a mistake on an issue, it would be polite to comment about it on that issue and not just create another issue.

schroeder-, 2022-09-20:
Fixed via #1040

closed 2022-09-20 by schroeder-

SharonBrizinov, 2022-09-21:
Thanks @schroeder- ! it looks good

kohtala, 2022-09-22:
Thank you. Trivy no longer complains with the new version.