FreeOpcUa/opcua-asyncio #1037: ServiceFault from server received  in response to CreateSessionRequest

omidfathi, 2022-09-13:
Hi, I try to connect an opc server, and when the program starts then this log appears and repeats to try connect to seerver:

ServiceFault (BadTooManySessions, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=None, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to CreateSessionRequest

```python
server_state = True
async def opcConnection(server_state, opcServer):
    try:
        if server_state:
            _SERVER_STATE = NodeId(ObjectIds.Server_ServerStatus_State)
            opc_url = opcServer
            client = Client(opc_url)
            print(client)
            await client.connect()
            client.session_timeout = 20000
            server_state = False
            return client, server_state
        else:
            print("opc_server connection Faild !!!")
    except:
        return None

async def main():  
    try:
        taskOPC = asyncio.create_task(opcConnection(server_state, opcServers[0]))
        client, server_state = taskOPC
    except:
        await main(clientMqtt)

if __name__ == "__main__":
    try:
        asyncio.run(main(clientMqtt))
    except KeyboardInterrupt:
        pass
    finally:
        print("Closing Loop")
```
[image]

AndreasHeine, 2022-09-13:
"BadTooManySessions" means the opc ua server does not have any free session available for you, so your server reached his sessionlimit!

omidfathi, 2022-09-13:
> "BadTooManySessions" means the opc ua server does not have any free session available for you, so your server reached his sessionlimit!

Yes, you are absolutely right, but I write this program to keep my program running and trying to connect to the server, even if it's shout down.
So, I start the server simulation and then run the program, everything is okay without any problem, but when i close the server completely, my program tries to connect to the server, after that i start the server again and this happens

AndreasHeine, 2022-09-13:
ok but a service fault is a response from a ua server !? so i am a little confused if you say the ua server is shutdown completely...

AndreasHeine, 2022-09-13:
and the runtimewarning is also not helpfull i checked the line and all close_secure_channel calles are awaited in master!

omidfathi, 2022-09-13:
> ok but a service fault is a response from a ua server !? so i am a little confused if you say the ua server is shutdown completely...
> 

Yes, when I shout down the server my client program stays in a loop and tries to connect to the server then I start the server and the client keeps repeating the opcConnection() function.

AndreasHeine, 2022-09-13:
could you make a wireshark trace of that behavior?

schroeder-, 2022-09-13:
First try to use the current pip version. Also make sure you call client.disconnect every time, even in case of exceptions. Also reduce the session timeout, to do this you have to call set_sessiontimeout before connecting.

AndreasHeine, 2022-09-13:
@schroeder- thats correct!

but i am wondering about the "BadTooManySessions" if you kill the server it does not matter with the old session anyways.

omidfathi, 2022-09-13:
> but i am wondering about the "BadTooManySessions" if you kill the server it does not matter with the old session anyways

I do it and now this errors happened:
[image]

omidfathi, 2022-09-13:
> 

you're right, it doesn't matter and I have no idea why this happens

schroeder-, 2022-09-13:
You have to update the library, because in our version are some bugs that prevent a correct disconnect.

schroeder-, 2022-09-13:
Ok I see the patch wasn't in 0.9.95. So best way is to install the master instead of pip:
pip install git+https://github.com/FreeOpcUa/opcua-asyncio.git

closed 2023-03-29 by schroeder-