FreeOpcUa/opcua-asyncio #1044: Version 0.9.96: Bad Request Too Large Error When Reading Value

sanssecours, 2022-09-21:
# Description

The latest version of the library seems to have problems when reading values from an OPC UA server. Code that worked fine with `asyncua` `0.9.95` now fails with a `TimeoutError`.

# Steps to Reproduce

1. Install [Docker](https://www.docker.com/products/docker-desktop/)
2. Store the following text in a file called `docker-compose.yaml` at a folder of your choice (e.g. `~/Downloads`):
   
   ```yaml
   version: "3.9"
   services:
     opcua:
       image: mcr.microsoft.com/iot/opc-ua-test-server
       command: "-p 4840"
       ports:
         - "4840:4840"
   ```

4. Enter the following command in the directory where you stored the file above

   ```sh
   docker compose up
   ```

5. Install `asyncua-0.9.96`:
   
   ```sh
   pip install asyncua==0.9.96
   ```

5. Copy the following code and store it as `test.py` 
   
      ```py
   from asyncio import run
   
   from asyncua import Client
   from asyncua.ua import NodeId
   
   
   async def test():
       async with Client(url='opc.tcp://localhost:4840') as client:
           node = client.get_node(NodeId('1:CC1001?Input1', 14))
           value = await node.read_value()
           print(f"Value: {value}")
   
   
   if __name__ == '__main__':
       run(test())
   ```

6. Run the script

   ```sh
   python test.py
   ```

# Expected Result

The script finishes successfully and prints the following output:

```
Value: 0.0
```

# Actual Result

The script fails with the following output:

```
Exception raised while parsing message from server
Traceback (most recent call last):
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/ua_client.py", line 86, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/common/connection.py", line 404, in receive_from_header_and_body
    return self._receive(chunk)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/common/connection.py", line 426, in _receive
    raise ua.UaStatusCodeError(ua.StatusCodes.BadRequestTooLarge)
asyncua.ua.uaerrors._auto.BadRequestTooLarge: "The request message size exceeds limits set by the server."(BadRequestTooLarge)
disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/rene/Downloads/test.py", line 15, in <module>
    run(test())
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/Users/rene/Downloads/test.py", line 8, in test
    async with Client(url='opc.tcp://localhost:4840') as client:
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/client.py", line 85, in __aenter__
    await self.connect()
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/client.py", line 277, in connect
    await self.open_secure_channel()
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/client.py", line 337, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/ua_client.py", line 304, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/site-packages/asyncua/client/ua_client.py", line 224, in open_secure_channel
    await asyncio.wait_for(self._send_request(request, message_type=ua.MessageType.SecureOpen), self.timeout)
  File "/Users/rene/.pyenv/versions/3.10.6/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

# Enviornment

Python Version: `3.10.6`
`asyncua` Version: `0.9.96`

# Additional Information

The script works as expected when I use `asyncua` `0.9.95`:

```sh
pip install asyncua==0.9.95
python test.py
# Value: 0.0
```

schroeder-, 2022-09-21:
We are working on a fix: #1043

sanssecours, 2022-09-21:
> We are working on a fix: #1043

Thank you 💖

oroulet, 2022-09-21:
Merged and release. Report if any issue

closed 2022-09-21 by oroulet